arduino-libraries/WiFi101 #2: UDP read not working on AVR boards

cmaglie, 2015-10-06:
The example UDPNTPClient returns always `0` as current timestamp.

This means that UDP send is working (because the time server receive the request) but the UDP read is not (because it reads always as `0x00`).

The bug appears only on AVR boards, probably the issue is due to a wrong assumption about datatype size on 8bit MCU.

mwernick1, 2015-10-30:
I have an arduino WiFi101 shield, and am experiencing this problem.  Is this an AVR device?  I tried increasing socket buffer, but no luck.  Any other ideas?

cmaglie, 2015-11-06:
@mwernick1 
may you try the patch here https://github.com/arduino-libraries/WiFi101/pull/5

cmaglie, 2015-11-20:
Not solved, there is still a buggy behaviour.
We are working on this one.

sandeepmistry, 2015-12-30:
This has been resolved with #7.

PR #5 (2015-11-20): Change read to return up to available bytes
Resolves #2.

Before read was only returning bytes that were buffered, instead of all the bytes in the parsed packet.

Also, changed available to return the bytes available in the current parsed packet, as per API spec. Previously, it was return bytes available in the buffer, if it was smaller than the packet size.

cc/ @cmaglie @facchinm
closed 2015-12-30 by sandeepmistry