arduino-libraries/WiFi101 #8: Problem using Arduino Wifi101 with UNO as a web server

sdumans, 2015-11-25:
I am using the sketch found at

      www.arduino.cc/en/Tutorial/Wifi101SimpleWebServerWiFi

The instructions on this web page are a bit messed up since they talk about lighting
an LED while the code is intended to write out the values of analog voltages.
However, as far as I can tell, that has nothing to do with the problem I am having
getting the sketch to work.

Here is my system configuration:

     Arduino Uno
     Arduino IDE 1.7.7
     Arduino Wifi 101 shield
     Linksys wireless router

I've also tried Arduino IDE 1.6.6 and that doesn't seem to make any difference.
Here is what I see in the Serial window when I run the code:

     Attempting to connect to SSID: linksys1
      SSID: linksys1
      IP Address: 192.168.1.108
      signal strength (RSSI):-50 dBm

When I then try to connect with 'http://192.168.1.108' with my web browser (firefox)
I then see

      new client
      client disconnected

in the Serial window.  This indicates to me that I am actually successfully communicating
with the WiFi 101 (I can also ping it ... another good sign) and that the problem is with
the code or that there is something else I need to do from my browser to get the code to
work.

agdl, 2015-11-26:
Well actually thank you because there was a bad link that now I fixed. The right example that must be there is 
https://github.com/arduino-libraries/WiFi101/blob/master/examples/SimpleWebServerWiFi/SimpleWebServerWiFi.ino

And previously there were:
https://github.com/arduino-libraries/WiFi101/blob/master/examples/WiFiWebServer/WiFiWebServer.ino

Using this last example if I upload it and I go to the ip address it prints on the serial monitor I can read out the analog values without problems. I also have firefox as browser. Have you something that maybe blocks the page in the browse? Haver you updated the WiFi101 library to the latest release?

sdumans, 2015-11-26:
Arturo,

Thanks for your note.  I was actually interested in reading the analog 
voltages from my Arduino.  That being
said,  I tried the LED sketch at the link you sent me and the same 
problem occurs.  Specifically, as soon as I
point my browser to the web address the result is the same as seen on 
the serial monitor:

         new client
         client disconnected

I do not see a message on my browser and I cannot control the LED.

As far as I know, I have the latest WiFi101 library ... how can I check 
that?  Also, please note that I am using
the WiFi101 shield with an Arduino Uno, not an Arduino Zero.

Steve Umans

Dr. Stephen D. Umans
Consultant
Life Fellow, IEEE
Member, US National Academy of Engineering

617-306-9773
umans@alum.mit.edu
http://alum.mit.edu/www/umans

agdl, 2015-11-26:
Argh... you are right. Actually it doesn't work with all the AVR boards! @sandeepmistry @cmaglie any idea?

We ( @facchinm and me) tried to solve it but without success

sandeepmistry, 2015-11-26:
I can confirm this behaviour with an Arduino Uno running the SimpleWebServerWiFi example with the master version of the library and using Chrome.

It might have something to so with the packet buffer sizes, as things work fine with the Zero.

agdl, 2015-11-26:
@sandeepmistry we tried to modify the buffer size but the problem persist

closed 2015-11-26 by cmaglie

sdumans, 2015-11-26:
> Closed #8 <https://github.com/arduino-libraries/WiFi101/issues/8> via

Thanks.  Does that mean that there is something I can to do in order to 
get around the problem?
Do I need to download another version of the IDE, a new library of some 
sort, etc.?

Tnx

Dr. Stephen D. Umans
Consultant
Life Fellow, IEEE
Member, US National Academy of Engineering

617-306-9773
umans@alum.mit.edu
http://alum.mit.edu/www/umans

agdl, 2015-11-27:
@sdumans we are planning a new release so you can update the library from the library manager. If you don't want to wait you can download the actual library as a zip here:

https://github.com/arduino-libraries/WiFi101/archive/master.zip

and replace the files in your sketchbook/libraries folder

sdumans, 2015-11-27:
Thanks Arturo.  I downloaded the zip file but quite frankly, as a 
newcomer to Arduiono, I don't know how to proceed to
update the WiFi101 library from the zip file.  Are there instructions 
somewhere?

Steve Umans

Dr. Stephen D. Umans
Consultant
Life Fellow, IEEE
Member, US National Academy of Engineering

617-306-9773
umans@alum.mit.edu
http://alum.mit.edu/www/umans

mbanzi, 2015-11-27:
Steve from the release number of your IDE I can see you're using the IDE release by somebody else. 

Please go to https://www.arduino.cc/en/Main/Software and download the 1.6.6 Arduino IDE 
then download the file  https://github.com/arduino-libraries/WiFi101/archive/master.zip
Unzip it and rename the folder to WiFi101 then follow this guide https://www.arduino.cc/en/Guide/Libraries at "Manual Installation"

sdumans, 2015-11-27:
Thank you!  I installed the library using the 'Sketch => Include Library 
=> Add .ZIP Library...' option and that worked.
The sketch now works properly and I am indeed able to control the LED 
from my browser.  I will now try some of
the other examples that I couldn't get to work such as reading the 
Analog voltages from my Arduino Uno board.

Steve Umans

Dr. Stephen D. Umans
Consultant
Life Fellow, IEEE
Member, US National Academy of Engineering

617-306-9773
umans@alum.mit.edu
http://alum.mit.edu/www/umans

jahangirali555, 2017-08-11:
@agdl

PR #9 (2015-11-26): Cast bit shifts for socket flags to be uint32_t
Resolves #8.

Without this change, the ```SOCKET_BUFFER_FLAG_SPAWN_SOCKET_MSK ``` and ```SOCKET_BUFFER_FLAG_PARENT_SOCKET_MSK``` were set to ```0``` because int size is 16 bits on AVR. This causes the return value of ```WiFiClient::connected()``` to report ``false```, after a new client was connected to the server.

Also, added another cast to remove a warning about unrelated bit shift in ```WiFiClient.cpp```.

cc/ @agdl @facchinm