arduino-libraries/WiFi101 #10: WiFi 101 Pin 5

sdumans, 2015-12-03:
The documentation for the WiFi 101 shield (www.arduino.cc/en/Main/ArduinoWiFiShield101) states
that "Digital pin 7 is used as a handshake pin between the WiFi Shield 101 and the board, and should not be used. "  I have been working with a motor-control system for which I am controlling the motor from a web-page.  It uses a number of digital output pins to control the motor.   My system consists of an Arduino Uno board with the WiFi 101 shield.  I have found that if I use Digital Pin 5, the system 
doesn't work properly but when I use Pin 7 there is no problem.  I am wondering if perhaps it is Digital pin 5 that "should not be used", at least for the Uno-WiFi 101 combination.

cmaglie, 2015-12-08:
@sdumans 
you're right pin 5 is used by the library to control the RESET of the wifi module. The documentation should be fixed ASAP.

soundanalogous, 2016-01-01:
I lost hours over this issue. Can someone please update the documentation and also confirm whether Pin 7 must be blocked as well or if it's only Pin 5. It's not clear from the comments above.

soundanalogous, 2016-01-02:
After more testing I can confirm that both pins 5 and 7 must not be used in the user's sketch.

facchinm, 2016-01-04:
Fixed in docs, thanks!

closed 2016-01-04 by facchinm