arduino-libraries/WiFi101 #11: no bytes available for client

bhmorse, 2015-12-09:
Hello. I am using the wifi 101 shield with an arduino zero, and running the web server example. When I try to curl to my board's ip address, sometimes I get the output of the GET request, but most of the time it just waits and stalls after printing out "new client". It seems that client.available() is returning zero, but curl claims the request has happened. Curls output is as follows:

curl -vL http://10.1.11.252/
*   Trying 10.1.11.252...
* Connected to 10.1.11.252 (10.1.11.252) port 80 (#0)
> GET / HTTP/1.1
> Host: 10.1.11.252
> User-Agent: curl/7.43.0
> Accept: */*
>

bhmorse, 2015-12-10:
Turns out if i curl with HTTP/1.0 it seems to be working. Ill investigate more

sandeepmistry, 2015-12-29:
@bhmorse could you please try with the latest master version of the library?

I'm wondering if #14 resolves this.

turkycat, 2015-12-31:
It may also be #16 that will resolve this, depending on the way the `WiFiClient` is being used. @bhmorse are you using a global variable for `WiFiClient`, or storing it in any other way? These are the same symptoms I was experiencing.

bhmorse, 2015-12-31:
I was using a global variable

turkycat, 2015-12-31:
Yep, this is very likely issue #15 (see example A) solved by PR #16

sandeepmistry, 2016-01-07:
@bhmorse #20 has been merged and should resolve this.

You can try the latest version from Github by downloading from [here](https://github.com/arduino-libraries/WiFi101/archive/master.zip) and adding to the IDE via: ```Sketch -> Include Library -> Add .ZIP Library```.

Please re-open if you still have issues.

closed 2016-01-07 by sandeepmistry