arduino-libraries/WiFi101 #17: WEP security does not work with WiFi101 library and MKR1000

turkycat, 2015-12-29:
Using a Linksys router configured with WEP wireless security & using the "ConnectWithWEP" example for simplest repro, each connection attempt with `WiFi.begin(ssid, keyIndex, key);` returns `WL_CONNECT_FAILED`. I have tried several keys and indices. I am able to connect to the network with my Android phone using any of the keys, and am able to ping the phone from my PC once connected. Let me know if you need me to investigate further or can suggest some things for me to try, as I have the setup here and ready to go.

I am using the latest code from today's updates, with my WiFiClient patch ( #16 ) on top.

As a side note, I have had no issues with open networks as well as all combinations of WPA/WPA2 security types using both AES and TKIP. These security types connect and have no issues exchanging large amounts of data with the device.

cmaglie, 2015-12-30:
/cc @ThibaultRichard

cmaglie, 2016-01-07:
Solved by #18. Feel free to reopen if the problem persists.

closed 2016-01-07 by cmaglie