arduino-libraries/WiFi101 #19: Use Wifi101 shield with Teensy 3.1

chensx2012, 2016-01-01:
I would like to use the Wifi101 shiled with Teensy 3.1 from pjrc.com, I have the following questions: (1) what hardware connections should I make between the shield and Teensy? (2) when I selected the Teensy board in Arduino IDE, the Wifi101 library was compiled with many different errors (many related with the _avr source), is it possible to add support with Teensy 3.1?

sandeepmistry, 2016-04-06:
Hi @chensx2012,

I've submitted pull request #53 to get this library working with a Teensy 3.1. Please see https://github.com/arduino-libraries/WiFi101/pull/53#issue-146406612 for the pin outs to use and the changes required to the current version of the library.

I've tested with the ScanNetworks and ConnectWithWPA examples.

cc/ @PaulStoffregen

chensx2012, 2016-04-08:
Thanks for your help, and now my Teensy 3.1 works with Wifi101. I have examined the scan networks and mac address. Later I found some other issues: 1) even after I connected a WPA wetwork, the WiFi.localIP() always return 0:0:0:0 (I checked the router management page, and saw the mac of the Wifi101 shield listed there), 2) for the firmware check example, it always showed Wifi shield: NOT PRESENT, and firmware version is "Err--"

sandeepmistry, 2016-04-12:
> 1) even after I connected a WPA wetwork, the WiFi.localIP() always return 0:0:0:0 (I checked the router management page, and saw the mac of the Wifi101 shield listed there)

I am not experiencing this. Do you have another Arduino board like the Uno to try this out with?

> 2) for the firmware check example, it always showed Wifi shield: NOT PRESENT, and firmware version is "Err--"

No issues with that example either, one note I have upgraded the f/w on my shield to 19.4.4. Here is the output I get:

```
WiFi101 firmware check.

WiFi101 shield: DETECTED
Firmware version installed: 19.4.4
Firmware version required : 19.4.4

Check result: PASSED
```

I would double check the wiring, I had to do it a few times when I setup things again today. Also, I am using the latest Teensy 1.28 release with Arduino IDE 1.6.8.

PR #53 (2016-04-14): Support for Teensy
Resolves #19.

Here are the pin connections I am using with the WiFi101 shield:

| WiFi 101 Shield  | Teensy 3.1 | Other |
| ---------------------- | -------------- | --------|
| IOREF | 3.3V | |
| 5V | | External 5V source |
| GND | GND | |
| MISO (ICSP header) | 12 (DIN) | |
| SCK (ICSP header) | 13  (SCK) | |
| MOSI (ICSP header) | 11 (DOUT) | |
| 8 (SS) | 8 |
| 7 (INTN) | 7 |
| 5 (RST) | 5 |

This is based on the [Arduino-WiFi101-schematic.pdf](http://www.arduino.cc/en/uploads/Main/Arduino-WiFi101-schematic.pdf).

@PaulStoffregen please review. Also, it would be great to [... cut]
closed 2016-04-14 by cmaglie