arduino-libraries/WiFi101 #22: WiFiClass::status() doesn't return WL_NO_SHIELD when shield is missing

andreimatei, 2016-01-09:
On my Arduino UNO board in a pristine state, with the WiFi101 shield *not* connected, the following program just hangs in that `status()` call. I've tried it with both versions 0.6 and 0.5 of the library.
I've tracked the hanging to `m2m_wifi_init() -> nm_drv_init() -> wait_for_firmware_start()`. This is probably expected, but I guess the question is how come `nm_drv_init()` does't fail earlier (in `nm_bus_iface_init()`?).

This has tripped me because the 101 shield I got was not particularly easy to plug in at first.

```
#include <WiFi101.h>

void setup() {
  Serial.begin(9600);
  while (!Serial) { }

  Serial.println("init");
  WiFi.status();
  Serial.println("why don't I get here?");
}
```

drp0, 2016-02-01:
I experience the same issue on an arduino mega R3 2560
I am waiting for the wifi101 shield to arrive and have been coding in anticipation.
With a bare board, status() fails.

No return from m2m_wifi_init(&param) in m2m_wifi.c
Last point of failure tracked to nm_read_reg(M2M_WAIT_FOR_HOST_REG) in nmasic.c (No return)
Naturally it should never have got this far into the initialise sequence.
As andreimatei above I expect that the nm_drv_init routine should return with an error much earlier!
David

sandeepmistry, 2016-02-01:
@drp0 I've submitted a patch in #23 to resolve this. Would you be able to try it out?

drp0, 2016-02-01:
Yes,
I tried clicking on #23 , then (541b5be#diff-57b01d2f3e2fef366d5fd9f83065c011L53).
This link opens part way into  src/driver/source/nmasic.c
I opened the view source, and replaced a copy of nmasic.c with the linked code-
It would not compile:
Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `chip_apply_conf'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `nm_clkless_wake'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `chip_idle'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `enable_rf_blocks'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `enable_interrupts'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `cpu_start'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `nmi_get_chipid'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `nmi_get_rfrevid'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `restore_pmu_settings_after_global_reset'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `nmi_update_pll'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':
[... 63 lines cut ...]
libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `nmi_get_otp_mac_address'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

libraries\WiFi101-master\driver\source\nmasic.c.o: In function `chip_apply_conf':

C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic.c:60: multiple definition of `nmi_get_mac_address'

libraries\WiFi101-master\driver\source\nmasic - Copy.c.o:C:\Users\laptop\Documents\Arduino\libraries\WiFi101-master\src\driver\source/nmasic - Copy.c:60: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Let me know if I have dived down the wrong rabbit hole,
David

drp0, 2016-02-01:
Test 2-
I replaced
 #define TIMEOUT						(0xfffffffful)
with
#define TIMEOUT						(0x2000ul)
Using the CheckWifi101FirmwareVersion sketch, the WIFI.status() routine returned in 15.22 seconds with NOT PRESENT (WL_NO_SHIELD)

David

sandeepmistry, 2016-02-02:
@drp0 great, thanks for trying. We'll have to wait for someone to review and merge #23.

drp0, 2016-02-02:
No problem,
I also tried #define TIMEOUT (2000)
as it seemed probable that the timeout was a milli-second value
The routine returned in ~3 seconds.

closed 2016-03-16 by agdl

z-l-p, 2017-12-15:
I regret to report that this problem is back in library versions 0.14 and up (2017.03.22). I tried with bare Mega 2560 and Pro Mini boards and reproduced same result as original post. 
(Confirmed fixed in 0.9 thru 0.13 though.)

sandeepmistry, 2017-12-18:
@z-l-p I've made pull request #205 to resolve this, it would be great to get your feedback on it.

z-l-p, 2017-12-18:
Thanks!  sandeepmistry's change works fine with my ATWINC1500 board (quickly returns correct values for present / !present). I can't do more in-depth testing because my board is now installed in an inaccessible place! Sorry.

PR #23 (2016-03-16): Change NMC1500 ASIC timeout for boot ROM and firmware start back to 2000
Resolves #22.

Timeout was changed from to ```0xfffffffful``` (from ```2000``` ) when support for firmware version 19.4.4 was added (https://github.com/arduino-libraries/WiFi101/commit/541b5be9993d03dde9ab3cd55c0c1b713984ad8f#diff-57b01d2f3e2fef366d5fd9f83065c011L53).

cc @ThibaultRichard