arduino-libraries/WiFi101 #25: Using WiFi101 shild in low power mode

Walter82, 2016-01-21:
Hello,
Is there a possibility to switch off the WiFi101 shield in software? For example: 
digitalWrite(5, LOW); - reduces the power consumption significantly (90%). After that I execute 
WiFi.init(); 
WiFi.begin(_ssid, _pass);
The WiFi connection is established. But:
WiFiClient client = WiFiClient();
client.connect(host, port); - is not working.

Maybe there is another possibility to switch off the WiFi101 shield in software?
Thank you very much and best regards, Walter

ThibaultRichard, 2016-01-22:
To shutdown WINC you have to put high chipen pin.
Beware to completely reinit wifi lib after doing that (it does a specific power up sequence with chipen and resetn)

Walter82, 2016-01-22:
>you have to put high chipen pin - done.
>Beware to completely reinit wifi lib after doing that - done.
WiFi.init();
WiFi.begin(_ssid, _pass); -> the wifi connection is there.
Problem: client.connect(host, port); - is not working.

agdl, 2016-06-07:
@Walter82 maybe you can be interested in https://github.com/arduino-libraries/WiFi101/pull/49

sandeepmistry, 2016-08-29:
This feature has been added in #49, which will be available in the next release.

closed 2016-08-29 by sandeepmistry