arduino-libraries/WiFi101 #30: Cannot form mqtt connections using WiFiSSLClient

InconceivableDuck, 2016-02-09:
The PubSubClient times out while attempting to connect. The underlying issue appears to be the .available() call always returning 0.

I've attempted to use the WiFi 101 shield on both the Zero and the Arduino 101 with the same results. Running [this example](https://github.com/arduino/cloud-examples/blob/master/mqtt/mqtt_send_digital/mqtt_send_digital.ino) as-is also yields the same issue.

Using non-SSL client connects successfully. I am running the latest master branch of the WiFi101 client.

sandeepmistry, 2016-02-12:
@InconceivableDuck I just tried [the example](https://github.com/arduino/cloud-examples/blob/master/mqtt/mqtt_send_digital/mqtt_send_digital.ino) you linked to. It works without any problems on my Zero + WiFi 101 shield (running firmware version 19.4.4).

What's the output of the [CheckWifi101FirmwareVersion.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/CheckWifi101FirmwareVersion/CheckWifi101FirmwareVersion.ino) example?

Does the [WiFiSSLClient.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/WiFiSSLClient/WiFiSSLClient.ino) example work for you?

InconceivableDuck, 2016-02-15:
Here's the result. I guess this could be the issue. I'll look into update the firmware.

```
WiFi101 firmware check.

WiFi101 shield: DETECTED
Firmware version installed: 19.3.0
Firmware version required : 19.4.4

Check result: NOT PASSED
 - The firmware version on the shield do not match the
   version required by the library, you may experience
   issues or failures.
```

InconceivableDuck, 2016-02-15:
WiFiSSLClient.ino yields:

```
HTTP/1.0 400 Bad Request
Content-Length: 54
Content-Type: text/html; charset=UTF-8
Date: Mon, 15 Feb 2016 22:38:02 GMT

<html><title>Error 400 (Bad Request)!!1</title></html>
disconnecting from server.
```

InconceivableDuck, 2016-02-15:
Rolling back to WiFi101 version 0.6 has resolved the issue. Thanks!

sandeepmistry, 2016-02-16:
@InconceivableDuck thanks for the info. I'm re-opening this and will change the title accordingly.

More info. on updating the firmware can be found here: https://github.com/arduino-libraries/WiFi101-FirmwareUpdater

sandeepmistry, 2016-03-16:
Resolved via #33.

closed 2016-03-16 by sandeepmistry