arduino-libraries/WiFi101 #36: Server uploads two images on a html page, the rest are not processed

drp0, 2016-02-25:
I have written a server application which works well, up to a point.
If a html page contains more than 2 pictures, only two are uploaded.
The missing pictures can be individual uploaded.
In IE they can forced by right click, show picture.
Refreshing the page does not alter the number of pictures uploaded.

[WiFi101server.txt](https://github.com/arduino-libraries/WiFi101/files/146992/WiFi101server.txt)

drp0, 2016-02-26:
Whilst looking for a work-round to this I discovered a library function that does not exist:
[https://www.arduino.cc/en/Reference/WiFi101](url): WiFi.getSocket()

Errror returned:

> Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\laptop\Desktop\Documents\Arduino\software\WiFi101server\WiFi101server.ino: In function 'void loop()':

WiFi101server:159: error: 'class WiFiClient' has no member named 'getSocket'

     Serial.println(client.getSocket() );

                           ^

exit status 1
'class WiFiClient' has no member named 'getSocket'
 
David

sandeepmistry, 2016-03-02:
@drp0 what board(s) are you testing with?

On the MKR1000, using the [SimpleWebServerWiFi.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/SimpleWebServerWiFi/SimpleWebServerWiFi.ino) example sketch and using ```nc <IP address> 80``` from a Mac. I can only establish 2 connections at a time, on the 2rd attempt Wireshark shows the connection being accepted, immediately followed by a disconnect.

sandeepmistry, 2016-03-02:
With the current ASF base, there is essentially a backlog of one pending connection until ```server.available()``` is called again. Any incoming connections will be accepted by the firmware, then immediately will be disconnected by the ASF code, see this block of code: https://github.com/arduino-libraries/WiFi101/blob/master/src/socket/source/socket_buffer.c#L216-L219

@ThibaultRichard any thoughts on how this can be improved?

drp0, 2016-03-03:
Hello Sandeep,

I am working with the Arduino mega 2560.

The wifi101 firmware is 19.4.4.

The library is version 0.8.0

 

Your experience on the Mac is like mine.

Could this use of a maximum of (up to) 2 sockets, also be the cause of the ftp download failure in #32?

 

The maximum number of tcp sockets in the library is more than two.

 

Cheers,

David

sandeepmistry, 2016-03-03:
@drp0 thanks for the info. The behaviour in this issue is specifically related to TCP server sockets, so unrelated to #32.

drp0, 2016-03-05:
What is the socket limit on the Amtel chip?
I have a memory of reading somewhere in the arduino library that the maximum number of tcp sockets was 7, although I can-not recall which file contained the information.
I had a quick look at the code snippit [ https://github.com/arduino-libraries/WiFi101/blob/master/src/socket/source/socket_buffer.c#L216-L219](url)
I am surprised it did not do a test of the number of sockets open, against the maximum number of sockets. Could this lead to a workround?
David

sandeepmistry, 2016-03-07:
> What is the socket limit on the Amtel chip?

Correct it's 7, see [socket.h](https://github.com/arduino-libraries/WiFi101/blob/master/src/socket/include/socket.h#L102).

> I am surprised it did not do a test of the number of sockets open, against the maximum number of sockets. Could this lead to a workaround?

Agreed there should be a check.

Only work around I can think of is to have your sketch call ```server.avaible()``` more frequently for now.

sandeepmistry, 2017-12-05:
Hi @drp0, if you are still interested in this, could you please try PR #204 when you get a chance. Thanks.

drp0, 2017-12-05:
Hello Sandeep-

 

you appear to have altered 19 files-

 

what is the simplest way to test this without manually editing all 19 files?

 

David

sandeepmistry, 2017-12-05:
Hi @drp0,

You can:

1) Backup your `<sketchbook>/libraries/WiFi101` folder
2) Download https://github.com/sandeepmistry/WiFi101/archive/socket-buffer-winc1500.zip
3) Extract zip
4) Place zip from library in  `<sketchbook>/libraries/WiFi101`
5) Restart the IDE and try it out

To restore, delete the `<sketchbook>/libraries/WiFi101` folder and replace with the one backed up in step 1).

drp0, 2017-12-05:
Hello Sandeep

 

I tried your library (Yes I do know how to test a library and restore the original)

 

I used my web server which had the issue that a page with more than 2 pictures only ever got 2, at most, back from the yifi101.

 

1)      I include  <TimeLib.h> as that worked in previous incarnations of yifi101

I seem to recall a lot of discussion on your choice of library for time support.

I get the warning “No system <time.h> header included, WiFi.getTime() wil always return 0”

So what have you changed and what are the implications for anything written in previous versions of YIFI101?

 

2)      The remote IP and port callback routine you gave me no longer works:

 

3)      // ************* Remote IP & port callback ******************************************

4)       

5)      static void myResolve_cb(uint8_t * /* hostName */, uint32_t hostIp) {

6)      WiFi._resolve = hostIp;

7)      }

8)       

9)      void mySocketBufferCb(SOCKET sock, uint8 u8Msg, void *pvMsg) {

10)   socketBufferCb(sock, u8Msg, pvMsg); // call the original callback

11)     if( u8Msg == SOCKET_MSG_ACCEPT) {

12)     tstrSocketAcceptMsg *pstrAccept = (tstrSocketAcceptMsg *)pvMsg;

13)     rip = pstrAccept->strAddr.sin_addr.s_addr;

14)     rport = pstrAccept->strAddr.sin_port;

15)     }

16)   }

17)    

18)   // ************* End Remote IP callback *********************************************

 

// ********* override the original callbacks **********

registerSocketCallback(mySocketBufferCb, myResolve_cb);

// ****************************************************

 

line 10 socketBufferCb appears to be no longer supported-

I hope you can maintain an option to get the remote IP for security reasons

 

3) I can now get 4 pictures back from yifi101 on one page- obviously a great improvement- further testing tomorrow

 

I have not yet tried to time the page serving rate- but it appears to be faster.

 

23.23 here in the UK,

 

Cheers,

David

sandeepmistry, 2017-12-06:
Hi @drp0,

Thanks for trying and apologies for not being clear on what I wanted you to test.

> 1)      I include  <TimeLib.h> as that worked in previous incarnations of yifi101
> I seem to recall a lot of discussion on your choice of library for time support.
>
> I get the warning “No system [...]

No changes have been made for this.

> 2)      The remote IP and port callback routine you gave me no longer works:

I've renamed `socketBufferCb` to `socket_cb`.

> I hope you can maintain an option to get the remote IP for security reasons

It's easier to make have a public API for this once the PR is merged.

> 3) I can now get 4 pictures back from yifi101 on one page- obviously a great improvement- further testing tomorrow
>
>  I have not yet tried to time the page serving rate- but it appears to be faster.

Ok, that's great news. The changes in the pull request allow the WiFi101 to have more than one accepted socket pending from a server. Previously as the issue is titled after one pending accepted connection, the WiFi101 library would close the socket.

drp0, 2017-12-06:
Hi Sandeep,
Good news indeed.
If the socket limit is 7, what would happen if a page contained more than 7 servable objects?

I tried  altering line 10 above to
socket_cb(sock, u8Msg, pvMsg);
and got 
socket_cb was not declared in this scope
Is there a temporary work round?

David

sandeepmistry, 2017-12-06:
> If the socket limit is 7, what would happen if a page contained more than 7 servable objects?

So one socket will be taken up by `WiFiServer`, so that leaves 6 left for incoming connections. You'll have to carefully manage `WiFiClient`'s and check if there is pending socket data if you are using an AVR based board.

> Is there a temporary work round?

In `src/WiFi.cpp` there the following line:

```
static void socket_cb(SOCKET sock, uint8 u8Msg, void *pvMsg)
```

remove the `static` so it becomes:

```
void socket_cb(SOCKET sock, uint8 u8Msg, void *pvMsg)
```

then in your sketch add `void socket_cb(SOCKET sock, uint8 u8Msg, void *pvMsg);` to forward declare it.

drp0, 2017-12-06:
That all works
So far the server appears more stable than before.
Thanks,
David

sandeepmistry, 2017-12-06:
Great, thanks for testing and the feedback :)

sandeepmistry, 2018-01-02:
Now that PR #204 is merged, this is fixed.

closed 2018-01-02 by sandeepmistry