arduino-libraries/WiFi101 #37: Documented Library function WiFi.getSocket() does not exist

drp0, 2016-02-26:
Whilst looking for a work-round to the server upload issue, I discovered a library function that does not exist:
[https://www.arduino.cc/en/Reference/WiFi101](url): WiFi.getSocket()

Errror returned:

> Arduino: 1.6.7 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

> C:\Users\laptop\Desktop\Documents\Arduino\software\WiFi101server\WiFi101server.ino:
> In function 'void loop()':
> WiFi101server:159:` error: 'class WiFiClient' has no member named 'getSocket'
>      Serial.println(WiFi.getSocket() );
> exit status 1

> 'class WiFiClass' has no member named 'getSocket'
 
David

agdl, 2016-02-26:
Hi! Thank you for pointing out this. Actually that page online is an error. The function does not exist and actually is quite useless so I will simply remove the online page and close this issue. Thank you!

closed 2016-02-26 by agdl

drp0, 2016-02-26:
Hi,

I found this in the include, socket.h:

*/

NMI_API sint8 setsockopt(SOCKET socket, uint8 u8Level, uint8 option_name,

       const void *option_value, uint16 u16OptionLen);

/** @} */

/** @defgroup GetSocketOptionsFn getsockopt

*  @ingroup SocketAPI

*   Get socket options retrieves

*    This Function isn't implemented yet but this is the form that will be released later.

*/

David

cmaglie, 2016-02-26:
@drp0
I cannot see how the original WiFi.getSocket() function could be used in sketches, it looks like more an internal method not intended to be used as a public API. Can you provide a working example of getSocket()? 

I'm not sure why it has been documented on the reference, IMHO it should be removed also from the old WiFi lib documentation, but maybe I'm missing something.

drp0, 2016-02-26:
Hi,

A working example of a non existant routine?

I note that the getSocket() link has now been removed from the WiFi101 library page https://www.arduino.cc/en/Reference/WiFi101

 

Has anyone had a look at #36 and can suggest a way forward?

 

Cheers,

David