arduino-libraries/WiFi101 #38: v0.8.0 does not work with the Uno + WiFi 101Shield

sandeepmistry, 2016-03-01:
The output of the ```ScanNetworks``` examples is:

```
MAC: F8:F0:5:F0:9A:85
Scanning available networks...
** Scan Networks **
MAC: F8:F0:5:F0:9A:85
Scanning available networks...
** Scan Networks **
MAC: F8:F0:5:F0:9A:85
Scanning available networks...
** Scan Networks **
MAC: F8:F0:5:F0:9A:85
Scanning available networks...
```

even though many access points are in range. Seems to be related to a change introduced in https://github.com/arduino-libraries/WiFi101/commit/f558ef50698aa891d188015904e1148afef44a45. Reverting to v0.7.0, the example is fine.

xarvey, 2016-03-07:
Same issue here, revert it back works 
https://forum.arduino.cc/index.php?topic=380311.0

sandeepmistry, 2016-03-07:
@xarvey I've submitted #39 to resolve this.

PR #39 (2016-03-16): Do not define PCINT* interrupt vectors on Mega and make non-weak on other AVR boards
 * Resolves #38
 * Reverts #29
 * Remove some AVR specific warnings

Tested with a WiFi 101 Shield + Uno, Leonardo, and Mega 2560

cc/ @facchinm
closed 2016-03-16 by agdl