arduino-libraries/WiFi101 #40: A call to get the remote IP in Server or AP mode would be very useful !

drp0, 2016-03-02:
This would help considerably with server security.
David

drp0, 2016-07-18:
I raised this way back in March?
It would be very surprising if Amtel had omitted to include a call to obtain the remote IP of the attached client in server/AP mode.
The lack of a library call to access the remote IP is a significant shortfall!

sandeepmistry, 2016-07-19:
Hi @drp0,

There is no API in the Atmel SDK to get the remote IP for a socket. However, when a server socket accepts a new connection the IP address and port of the client is provided via an event. So, it is possible to store this value for each socket. However, this would require 4 extra bytes per socket for each of the 7 TCP sockets available. If we also wanted to include the remote socket port, this would require another 2 bytes of storage per socket.

This would also introduce fragmentation to API's since the Arduino [Client interface](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Client.h) does not include support for retrieving the remote IP address of a connection. So, code that uses the new MKR1000 WiFiClient capabilities could not be used with other Client based network libraries.

drp0, 2016-07-19:
Hello Sandeep,
I understand the byte requirement. 
I am using a mega so can absorb the increased code size.
Which library routine could be used to intercept the  event? I would appreciate the opportunity of investigating the code potential.
David

sandeepmistry, 2016-09-06:
Hi @drp0,

Here's an example sketch of how to intercept the event:

```arduino
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "yourNetwork"; //  your network SSID (name)
char pass[] = "secretPassword";    // your network password (use for WPA, or use as key for WEP)

int status = WL_IDLE_STATUS;

WiFiServer server(23);

// new callbacks /////////////////////////////////////////////////////////
static void myResolve_cb(uint8_t * /* hostName */, uint32_t hostIp)
{
  WiFi._resolve = hostIp;
}

void mySocketBufferCb(SOCKET sock, uint8 u8Msg, void *pvMsg)
{
  socketBufferCb(sock, u8Msg, pvMsg); // call the original callback

  if( u8Msg == SOCKET_MSG_ACCEPT) {
      tstrSocketAcceptMsg *pstrAccept = (tstrSocketAcceptMsg *)pvMsg;

      Serial.println("Accepted connection from: ");
      IPAddress ip = pstrAccept->strAddr.sin_addr.s_addr;
      Serial.print(ip);
      Serial.print(":");
      Serial.print(pstrAccept->strAddr.sin_port);
      Serial.println();
  }
}
///////////////////////////////////////////////////////////////////////////

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // override the original callbacks
  registerSocketCallback(mySocketBufferCb, myResolve_cb);

  // start the server:
  server.begin();
}

void loop() {
  // wait for a new client:
  WiFiClient client = server.available();

  // ...

  
}
```

I'm going to close this issue for now.

closed 2016-09-06 by sandeepmistry

drp0, 2016-09-07:
Hello Sandeep,

 

This works perfectly.

I can access  the remote IP (and port) by storing them in global variables within the callback function:

then monitor the IP  following an if client call……..

WiFiClient client = server.available();

  if (client) {

……..

The client can be logged (and rejected if the ip is on a bad caller list)

 

Exactly as required.

 

It would be worth releasing this on the WIFI101 forum.

 

Thank you,

David

sandeepmistry, 2016-09-07:
Hi @drp0,

Thanks for the feedback. Feel free to post a link to this Github issue on the forum.

sandeepmistry, 2018-01-02:
New API's for this have been added in pull request #207, the changes have been merged and will be released shortly.