arduino-libraries/WiFi101 #44: v0.7.0 conflicts with Adafruit_VC0706/SoftwareSerial

GuidedByCthulhu, 2016-03-04:
I'm using an Uno r3, and trying to pair the wifi shield with the Adafruit TTL camera.  It appears that there is a little conflict going on however.  If I do this very simple sketch:

```
#include <Adafruit_VC0706.h>
#include <SPI.h>
#include <WiFi101.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

I get the following compiler errors:

```
libraries\WiFi101\bsp\source\nm_bsp_arduino_avr.c.o: In function `__vector_3':

\Documents\Arduino\libraries\WiFi101\src\bsp\source/nm_bsp_arduino_avr.c:65: multiple definition of `__vector_3'

libraries\SoftwareSerial\SoftwareSerial.cpp.o:C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:229: first defined here

libraries\WiFi101\bsp\source\nm_bsp_arduino_avr.c.o: In function `rx_pin_read':

\Documents\Arduino\libraries\WiFi101\src\bsp\source/nm_bsp_arduino_avr.c:60: multiple definition of `__vector_4'

libraries\SoftwareSerial\SoftwareSerial.cpp.o:C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:394: first defined here

libraries\WiFi101\bsp\source\nm_bsp_arduino_avr.c.o: In function `rx_pin_read':

\Documents\Arduino\libraries\WiFi101\src\bsp\source/nm_bsp_arduino_avr.c:60: multiple definition of `__vector_5'

libraries\SoftwareSerial\SoftwareSerial.cpp.o:C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial/SoftwareSerial.cpp:394: first defined here

collect2.exe: error: ld returned 1 exit status

```

Doing some quick googling, I see someone else had a similar problem here:
https://github.com/arduino/Arduino/issues/4534

However, they are attempting to do this with a mega (whereas I am using an UnoR3).  If I comment out the same lines mentioned in the link above, my wifi shield is no longer able to connect (no errors show up, it just simply fails to connect or scan for networks).  

I have not tested this with v0.8.0, as that version of the library appears to not work with the Uno r3.  The above bug report appears to have been closed, suggesting it has been fixed in v0.8.0?

xarvey, 2016-03-07:
Same issue here, I'm having conflict with < SoftwareSerial.h>, exact same error report.

Thanks for taking your time looking into this.

sandeepmistry, 2016-03-07:
@GuidedByCthulhu @xarvey 

Unfortunately on the Uno this library needs to use ```PCINT2_vect``` (vector 5) to listen to pin change interrupts on pin 7.

We can look into making the ```PCINT0_vect``` (vector 3) and ```PCINT1_vect``` (vector 4) vectors weak, but with the current technique of using the pin change interrupt on pin 7 ```PCINT2_vect``` (vector 5) cannot be made weak, so there will still be incompatibilities when trying to use this library with SoftwareSerial.

> I have not tested this with v0.8.0, as that version of the library appears to not work with the Uno r3. The above bug report appears to have been closed, suggesting it has been fixed in v0.8.0?

PR #39 is open to fix issues with the Uno not working in v0.8.0.

PR #29 only fixed SoftwareSerial support with the Mega in v0.8.0.

xarvey, 2016-03-07:
@sandeepmistry Thanks for your prompt reply, so does it mean that the wifi101 and software serial should work with Mega?

sandeepmistry, 2016-03-07:
@xarvey 

> does it mean that the wifi101 and software serial should work with Mega?

Yes this is correct, it is supported by v0.8.0 already (see #29).

xarvey, 2016-03-26:
@sandeepmistry I got the mega today and it works. Thank you so much.

sandeepmistry, 2016-03-28:
@xarvey perfect, thanks for confirming!

@agdl do you think we should change this to a documentation issue? Then mentioned that the Uno + WiFi 101 shield is not compatible with the SoftwareSerial library.

sandeepmistry, 2016-09-07:
@akash73 could we get someone to update the documentation with my comment in https://github.com/arduino-libraries/WiFi101/issues/44#issuecomment-202381172.

Then I think we can so the issue.

akash73, 2016-09-07:
ok we'll take care of it.

akash73, 2016-09-07:
Page updated
https://www.arduino.cc/en/Main/ArduinoWiFiShield101
I close the issue.

closed 2016-09-07 by akash73