arduino-libraries/WiFi101 #46: AP mode server sockets refuse connection after disconnecting and reconnecting to AP

sandeepmistry, 2016-03-07:
When using the [AP_SimpleWebServer.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/AP_SimpleWebServer/AP_SimpleWebServer.ino) example, the server socket is no longer in a listening state after a client disconnects from the AP and reconnects. WireShark shows a "connection refused".

Steps to reproduce:

1. Load the AP_SimpleWebServer.ino example sketch.
2. Connect device to created AP.
3. Open browser to http://192.168.1.1
4. Disconnect device from created AP.
5. Connect device to created AP.
6. Open browser to http://192.168.1.1

At step 6) the browser will not be able to establish a connection to the server. It looks like the WINC1500 firmware (19.4.4) is clearing server sockets after a client disconnects and re-connects to the AP.

sandeepmistry, 2016-03-07:
cc @ThibaultRichard

elitewill1200, 2016-08-02:
Anyone have any idea for a workaround for this? I was thinking something along the lines of pinging the default client IP, waiting till it gets something back, figure out when it leaves then restart the AP.  I couldn't get a ping through to the client however...

elitewill1200, 2016-08-02:
Oh, I should mention once you disconnect from the AP you can't connect to it again with a different advice. This might be related to the problems we are having.

elitewill1200, 2016-08-03:
I have discovered a workaround.  On #85, when AP changes to WL_AP_CONNECTED mode, make it start the socket. It will do this each time a device connects, like so

```
  if (status != WiFi.status()) {
    // it has changed update the variable
    status = WiFi.status();

    if (status == WL_AP_CONNECTED) {
      // a device has connected to the AP
      server.begin(); 
      Serial.println("Device connected to AP");
    } else {
      // a device has disconnected from the AP, and we are back in listening mode
      Serial.println("Device disconnected from AP");
    } 
  }
```

broddo, 2017-09-15:
I know this is a little old - but is using server.begin() really a good workaround? Surely that will just allocate a new socket each time its called - and once all the sockets are used up, it will stop allocating new ones. So this method could only be used successfully 7 times before causing problems. 

Would it make more sense to implement a stop() method in the WiFiServer class (similar to the ones already in the WiFiUDP and WiFIClient classes) that deallocates the current socket so that the begin() method can be called again?