arduino-libraries/WiFi101 #48: TLS 1.2 support advertised but no SHA-384/SHA-512 capability

facchinm, 2016-03-22:
_Moved from https://github.com/arduino/Arduino/issues/4730_
_@mashikawa  wrote:_

Hi,  I purchased the Arduino Wifi 101 shield for an IoT project and found that I could not connect to certain servers via SSL.  Some worked (www.google.com:443) and others did not (plot.ly:443).

Using Wireshark to monitor the TLS handshake I found that the WINC1500 hardcodes the TLS Client Hello value to 1.2 ( since firmware version 9.3.0 ), although the device does not handle ciphers longer than 256bits ( SHA-256 ).   This actually makes it only TLS1.1 compatible.  Some servers now implement SHA-384 and SHA-512 ciphers as part of the TLS1.2 protocol and they can respond to the TLS1.2 request with these ciphers, which the WINC1500 (Wifi101) cannot decrypt and so the connection fails.  I have recreated the issue using OpenSSL and have attached the output to this request for info.  

I raised a support request with Atmel ( Case 00037229 ), suggesting they allow the TLS value to be modified by the user via the software and they responded with the following message:

Created By: Anu Ramakrishnan (3/21/2016 4:09 AM)
[Recipients: Michael Kelsall]

Hi Micheal, 
Yes, we do not support SHA-384 and 512. Thank you for the feedback regarding the usage of TLS1.2 mode. We have raised a bug internally for this and we will follow up to get it fixed in future releases. 
We do not share firmware sources of the ATWINC1500, sorry for the inconvenience. 
Regards, 
Anu

This is a good response but I am now worried the issue may sit on a development list with low priority for an extended period of time.....   Can anyone help to progress this as I am sure the problem will get worse as more website implement longer ciphers to improve security.

Regards,
Michael.

[OpenSSL_ATWINC1500_TLS_Examples.txt](https://github.com/arduino/Arduino/files/183290/OpenSSL_ATWINC1500_TLS_Examples.txt)

mashikawa, 2016-03-22:
Cheers guys,  let me know if you want any help testing the fix.

mashikawa, 2016-03-30:
I received the following request from Atmel Support, so could "cmagile" please provide the incident numbers he referred to, so we can try to escalate this issue:

"Please respond above this email, avoid inline comments

Commented by Anu Ramakrishnan (Atmel) 2016-03-28 23:56 PDT
[Recipients: Michael Kelsall]

Hi Michael, 

I am following up on this internally. We will get back to you as soon as we have further information. 
Meanwhile, can you provide the case numbers/mail thread of the issues raised by the Arduino team in this regard, for our reference? 

Regards 
Anu "

From #4730:
cmaglie commented 8 days ago
For the record: we have already reported this problem to Atmel more than once in the last two months. Unfortunately there isn't much we can do except waiting for a new release of the firmware.

sandeepmistry, 2016-04-20:
Hi @mashikawa,

We've heard back from Atmel on this. SHA-384 certificate support is scheduled to be added in the June 2016 firmware release. Also, as of now, there are no plans to support SHA-512.

hiteule, 2016-06-26:
Hi,
Somebody has any news about the firmware update ? (Yep, we are in June 2016 :) )
Some site I want to use have a Comodo certificate with a SHA-384 encryption.

sandeepmistry, 2016-07-04:
Hi @hiteule,

I've contact Atmel support about this, unfortunately the June release is now pushed out to September.

cc/ @ThibaultRichard

q2dg, 2016-07-26:
In any case, if this issue succeed, please remember to update the line "Support TLS 1.1 (SHA256)" from https://www.arduino.cc/en/Main/ArduinoWiFiShield101 , please.

sandeepmistry, 2016-09-21:
@erhanalankus unfortunately nothing to report yet.

See: https://github.com/arduino-libraries/WiFi101/issues/90#issue-172510658:

> expected release data of late Oct 2016.

hiteule, 2016-11-16:
Hey, any news from Atmel about the firmware update ?

akash73, 2016-11-16:
postponed to January 2017

hiteule, 2017-01-17:
Someone have some news about de release date ?

akash73, 2017-01-17:
still remains the date of January.
we are all waiting for !

eliavgnessin, 2017-02-13:
Any news with this problem? It is urgent. Thx

mashikawa, 2017-02-13:
I gave up waiting for this a long time ago;  this was a simple fix, so it appears that Atmel are not  supporting this device.   About six months when I realised this was not going to be fixed,  I switched to ESP8266 board and added my own SHA-256 hashing to the IOT data uploads as a workaround.  Sounds like the new ESP-32 chip is worth checking out for secure uploads.

Regards,
Michael

facchinm, 2017-03-07:
FW 19.5.2 is finally out of the beta, you can check the ingtegration status here : https://github.com/arduino-libraries/WiFi101/pull/148

sandeepmistry, 2017-03-13:
Closing as #148 is now merged.

The 19.5.2 firmware is only available for model B of the WINC1500, this is used in the MKR1000 board.  Unfortunately, the WiFi101 shield uses model A, which Atmel has stopped supporting, so there is no 19.5.2 firmware release for it, 19.4.4 will be the latest firmware version that is compatible.

closed 2017-03-13 by sandeepmistry

sandeepmistry, 2017-03-13:
There's a PR build that contains the new firmware updater and the IDE available here for testing: https://github.com/arduino/Arduino/pull/6069#issuecomment-286223342.