arduino-libraries/WiFi101 #50: Arduino 101 + Wifi 101 Shield board freeze.

sandeepmistry, 2016-04-04:
Moved from https://github.com/arduino/Arduino/issues/4797

@apiascik:

> Hello,
>
>I'm working with the Arduino 101 (Curie) board and WiFi101 shield. I've been unsuccessful in getting even the ScanNetworks example to work. I can run the firmware detection example (it meets the firmware requirements), but execution freezes if I do anything more than that.
>
> This might be similar to [#4545]
>
> Observations
> 
> If I call Serial.begin, the first call to WiFiClass::scanNetworks() ultimately results in the board freezing. I observe no SPI traffic in my logic analyzer. If, however, I don't call Serial.begin, I continue to observe SPI traffic at intervals suggesting the sample is executing (scanning for networks).
>
> I dug into the WiFi driver code and found the following. I copied the m2m_hif.c's hif_send() function (called it hif_send2() ), and updated m2m_wifi.c's m2m_wifi_request_scan() function to call hif_send2() instead of hif_send(). I then added a return statement just before completing the packet transfer:
>
> ...
> reg = dma_addr << 2;
> reg |= (1 << 1);
> // return 0; <-- Doesn't freeze if I uncomment this.
> ret = nm_write_reg(WIFI_HOST_RCV_CTRL_3, reg); <-- Freezes if I let this line execute.
> if(M2M_SUCCESS != ret) goto ERR1;
> ...
>
> With the return I get no real results, as I'd expect, but the sketch continues to execute. I know it's not that line that causes the issue, but at this point, I'm unfamiliar with the driver as a whole, and don't know what what executes next. I'll keep digging.
>
> Configurations I've tried
>
> Windows 7 x64 USB2 and Windows 7 x64 USB3 (two different computers; applied the fix for USB 3)
> Arduino IDE 1.6.8
> Intel Curie Board Libraries 1.0.4 and 1.0.5
> WiFi 101 libraries 0.7.0, 0.8.0, and 0.9.0
>
> All combinations of the following devices:
> Arduino 101 boards:
> AE6642SQ55200CR
> AE6642SQ55201MY
>
> Wifi101 Shields:
> Neither of the 2 had serial numbers as far as I could tell. They were purchased and arrived within the last two weeks.
>
> Any help would be appreciated. Please let me know if I can provide any additional information.

sandeepmistry, 2016-04-05:
@apiascik I'm not seeing this issue at all with my Arduino 101 board using core v1.0.5 and a modified version of v0.9.0. I've tested with both firmware version 19.3.0 and 19.4.4 running on the shield.

The both the ```ScanNetworks``` and ```ConnectWithWPA``` examples are running fine.

The last two changes I mentioned in https://github.com/arduino-libraries/WiFi101/issues/51#issuecomment-205827647 are needed to build v0.9.0 on the 101, a regression was introduced in #47. How did you test v0.9.0 of the library?

Do you have another Arduino boards other than the 101 to test the shield out with?

apiascik, 2016-04-05:
@sandeepmistry 

> How did you test v0.9.0 of the library?

To try out the WiFi101 0.9.0 library, I removed WiFiMdnsResponder.cpp/h from the library folder. This is obviously not anywhere near a real solution, but it was enough to get things to compile so I could at least try out 0.9.0.

> Do you have another Arduino boards other than the 101 to test the shield out with?

I don't have any other Arduino boards, such as the Uno, to test things out with.

I haven't looked in depth at #51 (comment), but will try to update to 0.9.0 and (correctly) get things to compile to see if I observe any differences. I will report back on my results.

Is there anything else about my environment that might explain our different results? IDE 1.6.8? USB 2.0 vs 3.0? Please let me know if there's any other information I can provide.

FYI, I've been looking at the SPI traffic while attempting to scan networks. I don't see anything that indicates the code attempts to handle the M2M_WIFI_RESP_SCAN_DONE case (even when I don't call Serial.begin). I'm still familiarizing myself with the interface, so nothing is jumping out at me as incorrect/invalid.

apiascik, 2016-04-05:
@sandeepmistry

I installed WiFi101 library 0.9.0 and updated `WiFiMdnsResponder.cpp` and `pgmspace.h` with the modifications in to #51 and observed the same behavior. 

I used Arduino IDEs 1.6.8, [http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-4820-BUILD-546-windows.zip], and the latest nightly build (2016-04-05 12:13:10 GMT). Please let me know if I missed something. Just for completeness, this was on my USB 2.0-only system with Windows 7 x64.

Is there something else I can/should try?

facchinm, 2016-04-06:
Hi @apiascik ,
maybe the issue is not Curie or Wifi101 related but you are experiencing a side effect of an IDE bug. We are trying to solve it here https://github.com/arduino/Arduino/pull/4792#issuecomment-205908731, would you give this build a spin? Thanks you so much!

apiascik, 2016-04-06:
@sandeepmistry

Right now, the information I can get from debugging is limited. It would be helpful to have M2M_ERR print to the console, however, defining CONF_WINC_DEBUG results in compilation errors. What is involved in configuring the WiFi101 library for debug output?

apiascik, 2016-04-06:
@facchinm,

I tried build 547 but got the same result (at least from a serial output perspective). 

I just want to make sure I did the right thing to test the build. I downloaded the BUILD-547-windows.zip, extracted the files into that directory, and ran Arduino.exe right from there. After that didn't work, I pointed DeviceManager at the build folder but it indicated drivers were up to date. 

Please let me know if I should have done something more/differently (for example, should I have uninstalled 1.6.8 first)?

facchinm, 2016-04-06:
No need to uninstall the 1.6.8, all the steps you performed were correct. I'll keep you posted if I find something new

apiascik, 2016-04-06:
@facchinm, @sandeepmistry

I think I've identified another symptom that might help explain what's going on. I happened to have headphones plugged in and noticed what I considered to be an excessive number of USB disconnects/connects when executing the ScanNetworks example. I compared this to the Blink example, and sure enough, there was a difference.

I took a modified Blink example ( added "while(!Serial);Serial.println("starting");" ) with the Serial monitor window closed. Before opening it, the Serial Monitor device eventually appears. Opening up the Serial monitor window, the device stays there, and I hear no USB device disconnects.

On the other hand, starting from the same state with ScanNetworks, I observe the device disappear the "Scanning available networks...\n** Scan Networks ** text, the device disappears from Device Manager, then reappears a few seconds later. 

I tried closing and re-opening the SerialMonitor but got the message "Board at COM6 is not available". Hopefully this points toward something.

I'll try this on my other system tomorrow to see if I observe the same behavior.

facchinm, 2016-04-07:
@apiascik, I believe you are experiencing #4700 or #4777. The fix for these bugs has been merged in mainline, so if you downloads an hourly build from [here](https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip) the problem should disappear

apiascik, 2016-04-07:
@facchinm

Thanks for the quick feedback. I tried this release one my two systems. I observed the same behavior as before on both. The Serial Monitor device doesn't disappear until the network scan begins (after the MAC address has been retrieved). I don't observe this behavior when executing the CheckWiFi101FirmwareVersion.

apiascik, 2016-04-07:
@facchinm

I believe the "freeze" is the result of the `while(!Serial)` loop in setup(), i.e., the board (or sketch) appears to restart/reset.

I took the scanNetworks example, set up my own serial output `(SoftwareSerial mySerial(0,1);`, modified the output to distinguish the call to listNetworks in setup() from the one in loop() and observe the one in setup() being executed repeatedly.

[ScanNetworks2.ino.txt](https://github.com/arduino-libraries/WiFi101/files/208886/ScanNetworks2.ino.txt)

facchinm, 2016-04-08:
Ok, since the problem is not the serial port (or, at least, it's not THE problem), I believe that the last thing to check is that the board is not underpowered. This topic on the forum shows roughly the same problem (http://forum.arduino.cc/index.php?topic=377942.0) and it was resolved by using a wall charger.

apiascik, 2016-04-08:
@facchinm

Bingo! I didn't have a wall wart handy, so I tried different USB ports on my laptop and I have available networks printing out *sigh of relief*. I'll definitely be picking up a few adapters... Hopefully it'll be smooth sailing from here.

Thanks for all your help, and sorry it took so long to get to this conclusion!

facchinm, 2016-04-08:
:tada: :smile:

closed 2016-04-08 by facchinm