arduino-libraries/WiFi101 #51: v0.9.0 does not build on the Arduino 101

sandeepmistry, 2016-04-04:
```
/Users/smistry/Documents/Arduino/libraries/WiFi101/src/WiFiMdnsResponder.cpp: In member function ‘bool WiFiMDNSResponder::parseRequest()’:
/Users/smistry/Documents/Arduino/libraries/WiFi101/src/WiFiMdnsResponder.cpp:152:61: error: ‘memcmp_P’ was not declared in this scope
     if (memcmp_P(request, expectedRequestHeader, HEADER_SIZE) == 0 &&            // request header match
                                                             ^
/Users/smistry/Documents/Arduino/libraries/WiFi101/src/WiFiMdnsResponder.cpp:154:72: error: ‘strncasecmp’ was not declared in this scope
         strncasecmp(name.c_str(), (char*)requestName, requestNameLength) == 0 && // name match
```

1) The Arduino 101 does not provide ```memcmp_P``` in [avr/pgmspace.h](https://github.com/01org/corelibs-arduino101/blob/master/cores/arduino/avr/pgmspace.h)

2) There is no ```strncasecmp``` method.

The AVR and SAMD cores build fine, however SAM core also experiences 1).

@facchinm any thoughts on how to proceed with this? This breakage was introduced in  #47.

facchinm, 2016-04-05:
I believe that porting SAMD's `avr/pgmspace.h` to SAM and Curie will be the best long term solution :+1:

sandeepmistry, 2016-04-05:
@facchinm I've submitted the following pull requests:
 * https://github.com/arduino/Arduino/issues/4820
 * https://github.com/01org/corelibs-arduino101/issues/159
 * https://github.com/arduino-libraries/WiFi101/issues/52

Once all 3 are merged we can close this issue.

sandeepmistry, 2016-06-07:
The upstream changes have been merged and released (1.0.6 for the 101 core, 1.6.8 for SAM core).

Travis CI build is also now passing for the 101 and SAM.

cc/  @akash73

closed 2016-06-07 by sandeepmistry