arduino-libraries/WiFi101 #60: Wifi101 + Mega2560 + SD Card Breakout failure

GuidedByCthulhu, 2016-04-17:
I am having issues interfacing the Mega, Wifi Shield, and the Adafruit SD Breakout.  I am unable to use both the SD Breakout and the Wifi Shield concurrently; doing so produces failure states.  I have run the following three tests:

1) Wifi101 + Mega2560 Only (No wiring for SD Breakout)
*CheckWifi101FirmwareVersion works without issue / detects the shield

2) Adafruit SD Breakout + Mega 2560
CS to Pin 10
CLK to Pin 52
DO to pin 50
DI to pin 51
*CardInfo detects the card, and properly lists to contents of the SD card
`    Initializing SD card...Wiring is correct and a card is present.

    Card type: SD1

    Volume type is FAT16

    Volume size (bytes): 2030927872
    Volume size (Kbytes): 1983328
    Volume size (Mbytes): 1936

    Files found on the card (name, date and size in bytes):
    SYSTEM~1/     2015-07-03 10:40:12
      INDEXE~1      2015-07-03 10:40:12 76
    1.JPG         2000-01-01 01:00:00 48682
    2.JPG         2000-01-01 01:00:00 48768
    TRASH-~1/     2016-04-16 17:31:36
      INFO/         2016-04-16 17:31:36
        1JPG~1.TRA    2016-04-16 17:31:36 0
        1JPGTR~1.0OX  2016-04-16 17:31:36 57
      FILES/        2016-04-16 17:31:36
        1.JPG         2016-04-16 17:31:36 0
    3.JPG         2000-01-01 01:00:00 3700
    4.JPG         2000-01-01 01:00:00 3696
`

3) Adafruit SD Breakout + Wifi101 + Mega 2650 **(Failure State)**
CS to Pin 10
CLK to Pin 52
DO to pin 50
DI to pin 51
*CheckWifi101FirmwareVersion is unable to detect the shield.
*CardInfo detects the card, but is unable to list the contents of the card, and shows only the following output:
`    Initializing SD card...Wiring is correct and a card is present.

    Card type: SD1

    Volume type is FAT16

    Volume size (bytes): 2030927872
    Volume size (Kbytes): 1983328
    Volume size (Mbytes): 1936

    Files found on the card (name, date and size in bytes):
`

agdl, 2016-04-18:
Hi, the pin 10 is used as CS for the WiFi module. If I understood well you used the same pin for both the modules. If so you have to choose another pin for the SD card

GuidedByCthulhu, 2016-04-18:
You're correct;the Wifi101 uses Pin 10 as it's CS.  I didn't bother to review the Wifi101 details, as I assumed I remembered the correct CS pin. I've made a fool of myself; I do apologize for wasting your time.

sandeepmistry, 2016-04-18:
@GuidedByCthulhu no worries, great to hear things are working now.

closed 2016-04-18 by sandeepmistry