arduino-libraries/WiFi101 #61: How to stop AP?

javorosas, 2016-04-27:
There's a `beginAP` method, but how can I change back to STA mode? I want to connect to a WiFi network after receiving the ssid and password.

Thanks

sandeepmistry, 2016-04-27:
Hi @javorosas, this functionality is currently missing. However, I have submitted #62 to add it.

The patch includes a new ```WiFi.end()``` API to stop AP mode.

javorosas, 2016-05-02:
Looks great! Will it work the other way around? Changing from STA to AP mode?

sandeepmistry, 2016-05-02:
@javorosas thanks for trying!

> Will it work the other way around? Changing from STA to AP mode?

Yes, I have just tested it. Please let me know if it also works for you.

sandeepmistry, 2016-06-06:
This is closed now that @agdl has merged #62.

closed 2016-06-06 by sandeepmistry