arduino-libraries/WiFi101 #63: Cant even complile examples for wifi101 and arduino 101

a2liter, 2016-04-30:
When I try to compile the example sketch to scan for available networks I get the following error.

I am running IDE 1.6.9 hourly build 2016/04/28 4:33

The board I have selected is Arduino/Genuino 101

I am running library wifi101 0.9.0

I have read through all the forum posts I can find and I think I am doing all the suggested fixes for this issue.

**C:\Users\chris\Documents\Arduino\libraries\WiFi101\src\WiFiMdnsResponder.cpp: In member function 'bool WiFiMDNSResponder::parseRequest()':

C:\Users\chris\Documents\Arduino\libraries\WiFi101\src\WiFiMdnsResponder.cpp:152:61: error: 'memcmp_P' was not declared in this scope

     if (memcmp_P(request, expectedRequestHeader, HEADER_SIZE) == 0 &&            // request header match

                                                             ^

C:\Users\chris\Documents\Arduino\libraries\WiFi101\src\WiFiMdnsResponder.cpp:154:72: error: 'strncasecmp' was not declared in this scope

         strncasecmp(name.c_str(), (char*)requestName, requestNameLength) == 0 && // name match

                                                                        ^

Using library SPI at version 1.0 in folder: C:\Users\chris\AppData\Local\Arduino15\packages\Intel\hardware\arc32\1.0.5\libraries\SPI 
Using library WiFi101 at version 0.9.0 in folder: C:\Users\chris\Documents\Arduino\libraries\WiFi101 
exit status 1
Error compiling for board Arduino/Genuino 101.
**

akash73, 2016-05-02:
confirmed here, same behaviour.
first workaround is downgrade to 0.8.0 Wifi101 library version which works.

sandeepmistry, 2016-05-02:
I'm closing this as a duplicate of #51.

Now that https://github.com/01org/corelibs-arduino101/pull/159 has been merged, it will be resolved in the next 101 core release. Suggested work around for now is as @akash73 mentioned, downgrade to 0.8.0 of the library.

closed 2016-05-02 by sandeepmistry