arduino-libraries/WiFi101 #64: begin() with no encryption freezes

tigoe, 2016-05-09:
There seems to be a problem with begin() when it's got no encryption method.  

The following code loops continually if the AP does not allow a connection or isn't present:

````
char ssid[] = "my network";     
char pass[] = "gegegege"; 

void setup()
{
  Serial.begin(9600);
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to Network named: ");
    Serial.println(ssid);                   // print the network name (SSID);

    // Connect to WPA/WPA2 network:
    /status = WiFi.begin(ssid, pass);
    delay(5000);
  }
````

The following code, by contrast, does not loop, but freezes on the first call to begin():
````
char ssid[] = "my network";     

void setup()
{
  Serial.begin(9600);
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to Network named: ");
    Serial.println(ssid);                   // print the network name (SSID);

    // Connect to WPA/WPA2 network:
    /status = WiFi.begin(ssid);
    delay(5000);
  }
````

The network in question works fine with other networked devices, and has no encryption in place.

sandeepmistry, 2016-05-10:
I just tried this locally with a WiFi101 and MKR1000, no issue. Here are the sketches I was running:
 * [AP_SimpleWebServer.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/AP_SimpleWebServer/AP_SimpleWebServer.ino)
 * [ConnectNoEncryption.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/ConnectNoEncryption/ConnectNoEncryption.ino) with SSID changed to `wifi101-network`.

So one board is running in AP mode and the other connects to it.

Any thoughts on the best way to track down what you are seeing?

* What firmware version is on your board (see [CheckWifi101FirmwareVersion.ino](https://github.com/arduino-libraries/WiFi101/blob/master/examples/CheckWifi101FirmwareVersion/CheckWifi101FirmwareVersion.ino))?
* Are you using the latest version of this library (v0.9.0)?

tigoe, 2016-05-10:
I'm using the library version 0.9.1 according to my library manager. 

I used those same two sketches.
Haven't got my hardware nearby but I'll check FV when I do.

tigoe, 2016-05-10:
tried the same code snippets on three networks, and they worked on two, not on the third. Which is making me suspect the third network.

sandeepmistry, 2016-05-27:
Do you still access to this troublesome network? I can provide some instructions on some to enable debugging in this library to see if that yields any useful info.

tigoe, 2016-05-27:
Never managed to do it. But I decided the problem was most likely the network since it works fine everywhere else.

sandeepmistry, 2016-05-27:
Ok, I'm going to close this issue for now. We can re-open and spend more time on it if others have similar issues in the future.

closed 2016-05-27 by sandeepmistry