arduino-libraries/WiFi101 #65: How to renew IP Address (DHCP mode)?

PKGeorgiev, 2016-05-10:
Looking at the code I don't see a way to:

1) Detect if the IP Address lease has expired
2) Renew the IP address lease

I'm using MikroTik router with default lease time of 6 minutes (yeah, crazy). The the time eclipses, my MCU becomes unreachable via ping and cannot communicate with anything in my WiFi network.

PKGeorgiev, 2016-05-10:
I created separate issue #65 for IP Address renewal issues. It would be nice to have one function that report lease duration and another function to manually renew the lease.

sandeepmistry, 2016-05-27:
Hi @PKGeorgiev,

> Is there a way to:
> 
> 1) Detect if the IP Address lease has expired?
> 2) Renew the IP address if the lease has expired?

No, there are no API's for these.

I'm closing this in favour of #66, I think the library and/or WINC1500 firmware should take of this. We should keep things simple for library users :)

closed 2016-05-27 by sandeepmistry