arduino-libraries/WiFi101 #66: WiFi101 library does not renew MCU IP address when configured to use DHCP

PKGeorgiev, 2016-05-10:
When you use WiFi in DHCP mode it does not renew MCU's IP Address when the lease expires even trough WiFi.refresh() was periodically called.

sandeepmistry, 2016-05-27:
@PKGeorgiev I should have time to try to reproduce this next week.

However, since you've already done so based on your notes in #65, I think we should open a support ticket with Atmel about the firmware. Steps are:

1. Go to [this page](https://secure.atmel.com/design-support/)
2. Click "Open a Support Case"
3. Login and fill out the form (note the site is currently down for maintenance).
4. Cross reference the support ticket here.

This library is for the WINC1500 chip, so we need to mention that in the issue.

sandeepmistry, 2016-05-30:
@PKGeorgiev I tried to reproduce this today by setting up an access point with a DHCP lease time of 5 minutes. Things looked good, my MKR1000 sent another DHCP request after 5 minutes. I also used Wireshark to capture the traffic, and saw the request being sent out and a DHCP ack from my test router.

Could you please try a Wireshark capture to get more info.? My filter was set to ```udp && bootp``` to show only DHCP traffic.

PKGeorgiev, 2016-05-30:
This is very strange. Looking at WireShark I can see that MCU is constantly requesting DHCP lease renew. In Mikrotik's ghcp logs I also see that the router is sending DHCP ACK to first few requests. But the MCU does not react on these ACKs. 
I will try with a different router to see if there is a difference. I'll keep you informed.

PKGeorgiev, 2016-06-09:
@sandeepmistry I opened case at Atmel (00040760) and did some tests. It seems that the problem is related to MikroTik. I tried with TP-Link router and MKR1000 does renew its DHCP lease. I will contact MikroTik's support to try to resolve the problem.
If you want keep this issue opened until I get some results from MikroTik's support. It would be helpful to MikroTik & MKR1000 owners.

sandeepmistry, 2016-06-09:
@PKGeorgiev ok, thanks for the update, I've renamed the issue accordingly.

matwho, 2016-08-30:
I have seen exactly the same problem with DHCP renew MKR1000 (also tried Adafruit Feather MO WiFi) and a MikroTik router. Using a static IP until Atmel give us a fix.

sandeepmistry, 2016-09-06:
Hi @matwho, I'd suggest you also open a support case with Atmel to let them know about your issues. There are some instructions in https://github.com/arduino-libraries/WiFi101/issues/66#issuecomment-222257053.

luistope, 2017-02-17:
Same issue with end of lease time. M2M_WIFI_REQ_DHCP_CONF event never received. 
Open socket is not possible without IP "assigned".  
In the library is not available the function m2m_wifi_request_dhcp_client(). It's obsolete and empty. 
So we can not reconnect again to the network after this case leaving the device unconnected and lost.
Any update about this??

sandeepmistry, 2017-03-13:
@PKGeorgiev any news on Atmel/Microchip on this?

There's PR build of the IDE that contains the new v19.5.2 WINC1500 firmware: https://github.com/arduino/Arduino/pull/6069#issuecomment-286223342. Any one up for testing it with the master version of the library to see if this has been fixed?

matwho, 2017-03-14:
After a quick test [Arduino#6069](https://github.com/arduino/Arduino/pull/6069) seems to have fixed my problems with DHCP and MikroTik router.

sandeepmistry, 2017-03-14:
@matwho awesome! Thanks so much for trying it out!

closed 2017-03-14 by sandeepmistry