arduino-libraries/WiFi101 #69: CheckWifi101FirmwareVersion unhelpful message

PaulStoffregen, 2016-06-04:
This example should print a more helpful message, at the very least with a link to the update utility.  Ideally, it should also print a brief summary of how to accomplish the update.

Currently, the end-user experience in the case of a too-old firmware is this:

    Check result: NOT PASSED
     - The firmware version on the shield do not match the
       version required by the library, you may experience
       issues or failures.

That's pretty far from the ease-of-use reputation Arduino has earned over the years.  The Wifi101 pages don't even mention this example or how to upgrade:

https://www.arduino.cc/en/Main/ArduinoWiFiShield101
https://www.arduino.cc/en/Reference/WiFi101

Good usability in error messages requires not only informing users of a problem, but giving them instructions or a link to where they can find easy-to-follow instructions to solve the problem.

PaulStoffregen, 2016-06-04:
Likewise, this library has a FirmwareUpdater example, which contains only a LPGL license in its comments.  Absent is the usual Arduino style of human readable explanation of what the example does, how to use it, and what other stuff is necessary.

agdl, 2016-06-21:
@PaulStoffregen please see the reference in this issue and tell me if it is ok for you!

PaulStoffregen, 2016-06-21:
That's certainly an improvement over no link at all.

But imagine you're using the Wifi101 for the first time and you get that message.  So you go to that page.  Maybe you're ok with the description saying it's about certificates and you figure out it also does the firmware upgrade.

So you try to follow the 4 step process to update your firmware.  Let's assume you know how to use command line (many people do not).  Where are you supposed to get "winc1500-uploader"?

Are you able to imagine a person trying to follow these steps?  Or maybe you don't care?  Really, Wifi101 isn't a product I sell or support.  It's up to you whether this is a usability or quality issue.

YOU (not I) need to decide whether these improved the on-screen documentation and step-by-step instructions are up to Arduino's quality standards.

agdl, 2016-07-26:
@PaulStoffregen have a look at this 
https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin

PaulStoffregen, 2016-07-26:
Wow, looks nice!

agdl, 2016-09-07:
@PaulStoffregen do you like the new plugin? What about closing this issue? :)

closed 2016-09-07 by PaulStoffregen