arduino-libraries/WiFi101 #72: SimpleWebServerWifi hangs after a time in MKR1000.

RamonRibes, 2016-06-18:
I intended control a home device using a MKR1000 and a web server from your Wifi101.
So i did, but the board hangs after an unpredictable time and stops responding client's requests.
In order to find the problem, i loaded the example file SimpleWebServerWifi, and tested it alone. I just changed:
-My wify name&password.
-The pin number 9 to 6 (it's the built-in led of MKR1000).
Well, i found this sketch hangs since a few minutes after uploaded, to a maximum of 2 days.
I exposed this issue in the MKR1000 forum, and at least one other guy got same behaviour.
Same issue is found with the sketch WiFiWebServer, so i think it's related to the webserver code.
I have 8 MKR1000 boards, and checked this issue in all them.
Don't know if at the end, it's hardware or software related, but would be helpfull if someone else could test by itself and report the results.
Thanks!

sandeepmistry, 2016-06-20:
Hi @RamonRibes,

I tried to reproduce this issue with my MKR1000 last week, however encountered no issues after running `SimpleWebServerWifi` using pin 6 for the LED for 5 days. After this I needed to use my MKR1000 for something else, so had to disconnect it.

Could you please provide some more details on your setup:
 * How were you powering the MKR1000? (I was using a PC)
 * How frequently were you trying to access the web server on the MKR1000? What browser were you testing with?
 * Did you try to ping your MKR1000 when the web server stopped responding?

There's also the possibility that the WiFi lost connection for whatever reason. The `SimpleWebServerWifi` example sketch does not have any logic to re-connect WiFi once it is disconnected.

RamonRibes, 2016-06-20:
Hi, @sandeepmistry, thanks for responding!
The answer to your questions follows:
-Boards are powered by: mobile phone charger to micro-USB input, PC USB port to micro-USB input, and a power supply connected to pins VIN (+) / GND (-).
-After uploading the sketch, i check if the web server is started (commonly it does, but sometimes it fails), and then i re-check 5-6 times per day. Webbrowser is Chrome (PC & mobile versions).
-Yes, and it fails, because there is no internet connection with the board. I use a mobile application to search for wifi-connected devices, and realized when a MKR1000 stops responding, it also dissapears from the wifi's device list.

Additional comments:
1. In my last series of tests, i kept 4 boards working with same sketch. 3 of them must be re-started many times, in a variable times, but the fourth, kept working for 3 days (now still working), powered by a mobile phone charger.
2. The message given by the browser when a fail occurs is either:
  -Error: Empty response from server.
or,
  -Error: Address is unreachable.
3. I tried other sketch to avoid wifi unconnections. This is the part:

> void loop() {
  reconnectWifiIfNecessary();
  if (noIpCheckEnabled == true) {
    if (isTimeToCheckIP()) {
      checkForVerifyIpShortAlarm.setAlarm(initialTimeToCheckIP);
      Serial.println("Checking No-IP...");
      checkAndRenewIP();
      Serial.println("No-IP checked!");
    }
  }
  // Sending AC webpage:
  processAcServer();
  delay(200);
}
void reconnectWifiIfNecessary() {
  while ( WiFi.status() != WL_CONNECTED) {
    Serial.println("Attempting to reconnect wifi...");
    // Connect to WPA/WPA2 network:
    WiFi.begin(ssid, wifiPass);
    // wait 20 seconds for connection:
    delay(20000);
    if (WiFi.status() == WL_CONNECTED) {
      Serial.println("Wifi reconnected!");
    }
  }
}

> 

4. My environment is as follows:
-8 Arduino MKR1000 boards.
-Arduino IDE 1.6.9.
-Wifi101 library vers. 0.9.1.
-Windows 10 Home Edition.
5. I'm pretty sure i do something wrong, because isn't possible so many people doesn't have this issues. The thing that surprised me is, when i tried to fix the problem, is to find the issue also in the example sketch!. I reported it in an Arduino Forum, and someone else observed the same behaviour (my username there is 'mephala'):
http://forum.arduino.cc/index.php?topic=376619.msg2796935#msg2796935
http://forum.arduino.cc/index.php?topic=401107.15
6. Excuse the lot of time you're spending with my problem. Also my english...

Thanks!

RamonRibes, 2016-06-23:
The board who worked for 5 days was disconnected to use the phone charger. After the charge, i connected it again, and failed responding in few minutes.
I don't what to do now, nor what to look at...

sandeepmistry, 2016-06-27:
Hi @RamonRibes,

Thanks for the updates. Unfortunately unless we have a consistent way to reproduce this, it's going to be very tough to solve.

I would suggest the following:
 * See if we can determines if this is caused by either: time the server is up OR the number of requests the server receives. For example, we try 100 back to back requests in a row in a short period of time.
 * Add more logging to the sketch, and connect the MKR1000 to a PC and monitor if the sketch gets stuck anywhere.

RamonRibes, 2016-07-04:
Hi,
In response to your request, i tested this sketch:
`/*
  WiFi Web Server LED Blink

 A simple web server that lets you blink an LED via the web.
 This sketch will print the IP address of your WiFi Shield (once connected)
 to the Serial monitor. From there, you can open that address in a web browser
 to turn on and off the LED on pin 9.

 If the IP address of your shield is yourAddress:
 http://yourAddress/H turns the LED on
 http://yourAddress/L turns it off

 This example is written for a network using WPA encryption. For
 WEP or WPA, change the Wifi.begin() call accordingly.

 Circuit:
 * WiFi shield attached
 * LED attached to pin 9

 created 25 Nov 2012
 by Tom Igoe
 */
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "xxx";      //  your network SSID (name)
char pass[] = "xxx";   // your network password
int keyIndex = 0;                 // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
WiFiServer server(80);

void setup() {
  Serial.begin(9600);      // initialize serial communication
  pinMode(9, OUTPUT);      // set the LED pin mode

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    while (true);       // don't continue
  }

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to Network named: ");
    Serial.println(ssid);                   // print the network name (SSID);

    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);
    // wait 10 seconds for connection:
    delay(10000);
  }
  server.begin();                           // start the web server on port 80
  printWifiStatus();                        // you're connected now, so print out the status
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("new client");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> turn the LED on pin 9 on<br>");
            client.print("Click <a href=\"/L\">here</a> turn the LED on pin 9 off<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          }
          else {      // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        }
        else if (c != '\r') {    // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(9, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(9, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("client disonnected");
  }
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
  // print where to go in a browser:
  Serial.print("To see this page in action, open a browser to http://");
  Serial.println(ip);
}`
It's from the example 'SimpleWebServerWifi'. Only changed my ssid name & password.
Test procedure:

Board one:
Connected to pc-usb cable, and uploaded the sketch.
Disconnected from pc-usb cable.
Connected to a stand-alone mobile charger.
Accessing through mobile/pc via wifi. No issues after 1 hour.

Board two (first test):
Connected to pc-usb cable, and uploaded the sketch.
Kept connected to pc-usb cable.
Open serial monitor.
Accessing through mobile/pc and monitoring messages via serial monitor (many times).
Closed serial monitor.
Request fails on the next access (browser is waiting).
-If i send a few more access requests, board stops responding, even if i re-open the serial monitor.
-If i re-open the serial monitor immediately, it goes on responding to serial and browser.

Board two (second test):
Connected to pc-usb cable, and uploaded the sketch.
Kept connected to pc-usb cable.
Serial monitor kept closed.
Accessing through mobile/pc via wifi. No issues after 10 minutes.
Board stops responding. Chrome displays 'ERR_EMPTY_RESPONSE' code. Microsoft Edge says can't access the web page. Board is still responding to pings.

Thanks for your attention!

sandeepmistry, 2016-07-04:
Hi @RamonRibes,

Thanks for the info. on how to reproduce. I've been able to using a Windows 10 VM, by opening and closing the Serial monitor. Previously, I was using OS X which does not appear to have this issue.

I'll post any updates here.

sandeepmistry, 2016-07-04:
Hi @RamonRibes,

I've submitted a pull request to the SAMD core to resolve the web server sketch not responding when the Serial monitor is closed. You can try out the patch by following these instructions: https://github.com/arduino/ArduinoCore-samd/pull/152#issuecomment-230345077

I'm going to setup a MKR1000 with a USB power adapter next.

RamonRibes, 2016-07-06:
After following your instructions to incorporate the patch #152, i'm afraid to inform this. The above test 'Board two (first test)' shows exactly the same behaviour :-(

sandeepmistry, 2016-07-11:
Hi @RamonRibes,

Did you select the right board as per step 5? Could you please share a screenshot of the board you selected? My colleague @facchinm tested this for me on a Windows XP PC and it worked for him.

I've also submitted #77 to try to resolve the `ERR_EMPTY_RESPONSE` condition you were seeing before.

I still haven't been able to reproduce the issue you are seeing when connect to a USB power adapter. I have MKR1000 connected to a USB power adapter since last week, and I can still connect to the web server without any issues.

RamonRibes, 2016-07-12:
Hi,
Attached to this post you'll find a screen capture, as requested.
[image]
My next steps:

- [ ] Test #77 

- [ ] Test SimpleWebServerWifi in Arduino IDE **in WindowsXP compatibility mode** (Still no chance to test it in another computer).

Would you suggest another tests?

regards.

sandeepmistry, 2016-07-12:
Hi @RamonRibes,

Thanks for the screenshot, it looks good. Maybe you can also try restarting the IDE and uploading the sketch with that board selected.

I've mentioned another more simple test you can run in: https://github.com/arduino/ArduinoCore-samd/pull/152#issuecomment-232076896 (since is for opening and closing the serial monitor freezing the MKR1000).

Otherwise, your next step list looks good. Thanks for your help tracking down these issues!

RamonRibes, 2016-07-14:
Tested SimpleWebServerWifi in Arduino IDE in WindowsXP compatibility mode, and also in administrator mode. No changes found...

nelsballs, 2017-04-25:
Are there any updates on this one ? I encounter exactly the same issue as @RamonRibes  on my MKR1000. A simple connection to a (working) server every 30 seconds with a check to ensure Wifi connection but the MKR1000 is unable to connect after a couple of minutes to a couple of hours.

RamonRibes, 2017-04-26:
nelsballs,
i'm afraid to inform that, since my last post, i switched using ESP8266 module instead of MKR1000.
Was a little bit hard changing the code to the new board, but is worth it (...and cheaper, also!).
Now, more than 7 mounths later, the new board is working properly, without any issue.
Nonetheless, it would be very happy if this issue was solved: i still carry 8 MKR1000 boards i bought!

mamama1, 2017-04-26:
also try ESP32, it has dropped to about 10€ now.
arduino software for winc1500 based boards is very crappy unfortunately. i don't understand how it is possible that the software is still so bad after that period of time. must be some issue with the hardware too. i've burnt 50€ with two winc1500 modules too and won't be using them anymore because esp8266 and esp32 work like a charm.

nelsballs, 2017-04-26:
Thanks guys for your advices ! I will definitely consider esp8266 and esp32 for the next experimentations. It's indeed a shame to not tackle this MKR1000 issue : /

sandeepmistry, 2017-05-02:
Hi Everyone,

We haven't looked further into the original issue with f/w version 19.4.4 as it couldn't be reproduced with our boards and network.

Has anyone checked if the latest 19.5.2 firmware release from Microchip/Atmel still has this issue. You'll need to install the latest version of the WiFi101 library. Instructions on how to update the firmware on the board can be found here: https://www.arduino.cc/en/Tutorial/FirmwareUpdater

RamonRibes, 2017-05-14:
Well, thanks to sandeepmistry for keeping the issue alive!
In response to your request, i did the following actions:
1. Updated firmware on MKR1000 boards as instructed.
2. Updated wifi101 library.
3. Tested example file 'SimpleWebServerWiFi', after editing my ssid&passw and led pin (9 to 6).
4. Test was ok but, when wifi is lost, the board doesn't re-connects automatically, and keeps unconnected (is this an issue to fix?).
5. Modified the test sketch to test connection and auto-reconnect. Also to account re-connections.
6. The test was ok, but sometimes hangs stuck waiting client's entry. So i added a timeout routine to exit (also to count them).
7. The test is doing ok for a week on 4 boards simultaneously. In this period, i accessed aprox. 100 times, having aprox 9 timeouts and 2 re-connections per each of the boards tested.
8. The only issue i noticed is, when accessing the board from outside the home router, sometimes the communication fails and the timeout routine acts, allowing you access in the netx attempt. This never happend when accessing from the local network (to the private IP address). I don't qualify it as a problem.
The sketck i used is available to any who asks for it.
So, sandeepmistry, i consider this issue as SOLVED. Thank you very much!

sandeepmistry, 2017-05-16:
@RamonRibes thanks for trying the new firmware release!

> the board doesn't re-connects automatically, and keeps unconnected (is this an issue to fix?).

Yes, this is the expected behaviour, #108 is open to the track auto re-connect feature.

I'll close this issue for now, since things seem solved. Please open a new issue for any problems you encounter with the 19.5.2 firmware. Thanks.

closed 2017-05-16 by sandeepmistry

larstobi, 2017-05-26:
@RamonRibes Yes, thank you, I'd very much like to see how you did it in your sketch. You'll find my email address on my account page https://github.com/larstobi

Thanks!

RamonRibes, 2017-05-26:
@larstobi here you have the sketch:
`
/*
  WiFi Web Server LED Blink

  A simple web server that lets you blink an LED via the web.
  This sketch will print the IP address of your WiFi Shield (once connected)
  to the Serial monitor. From there, you can open that address in a web browser
  to turn on and off the LED on pin 6.

  If the IP address of your shield is yourAddress:
  http://yourAddress/H turns the LED on
  http://yourAddress/L turns it off

  This example is written for a network using WPA encryption. For
  WEP or WPA, change the Wifi.begin() call accordingly.

  Circuit:
   WiFi shield attached
   LED attached to pin 6

  created 25 Nov 2012
  by Tom Igoe
*/
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "mySSID";       //  your network SSID (name)
char pass[] = "myPassword";   //  your network password

IPAddress privateIP(192, 168, 1, 61); // It's easyer to access a static IP!
IPAddress dns(192, 168, 1, 1);
IPAddress gateway(192, 168, 1, 1);
IPAddress netMask(255, 255, 255, 0);

int accessCount = 0;                      // will count page access.
int connectionCount = 0;                  // will count wifi re-connections.
int forcedExits = 0;                      // will count timeouts in the 'while' loop.
byte lineCounter = 0;                     // Used to format serial output.
WiFiServer server(80);

void setup() {
  Serial.begin(9600);      // initialize serial communication
  pinMode(6, OUTPUT);      // set the LED pin mode
}

void loop() {
  WiFiClient client;
  char activeTimeString[60];
  int sec = millis() / 1000;
  int mn = sec / 60;
  int hr = mn / 60;
  int days = hr / 24;

  verifyWifiConnection();
  if (++lineCounter >= 100) {
    lineCounter = 0;
    Serial.println(".");
  }
  else {
    Serial.print(".");
  }
  client = server.available();   // listen for incoming clients
  snprintf(activeTimeString, 60,"%d days, %d:%02d:%02d hours.", days, hr % 24, mn % 60, sec % 60);
  if (client) {                             // if you get a client,
    Serial.println();
    Serial.println("new client");             // print a message out the serial port
    String currentLine = "";                  // make a String to hold incoming data from the client
    unsigned long actualTime = millis();      // get time of loop entry.
    while (client.connected()) {              // loop while the client's connected
[... 77 lines cut ...]
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
  // print where to go in a browser:
  Serial.print("To see this page in action, open a browser to http://");
  Serial.println(ip);
}

void verifyWifiConnection() {
  // attempt to connect to Wifi network:
  while ( WiFi.status() != WL_CONNECTED) {
    Serial.print("Attempting to connect to Network named: ");
    Serial.println(ssid);                   // print the network name (SSID);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    WiFi.config(privateIP, dns, gateway, netMask);
    if (WiFi.begin(ssid, pass) == WL_CONNECTED) {
      ++connectionCount;
      printWifiStatus();                        // you're connected now, so print out the status
    }
    // wait 10 seconds for connection:
    delay(10000);
    server.begin();                           // start the web server on port 80
  }
}
`

alexmoskos, 2017-08-17:
Just commenting to validate one suggestion in the conversation. I experienced similar problems where the MRK1000 would hang after 10-120 seconds after turning on the LED a couple times, resetting the MKR1000 would only allow it to work for another 0-2 minutes. After updating Firmware to WINC1501 Model B 19.5.2, everything is working great. I am able to disconnect power, power it back on, and have it resume working. 50+ connections using the web to turn on the LED.  Seems the FW update did the trick

henrikkoch, 2017-09-11:
Just update the WiFi101 firmware according to this: https://www.arduino.cc/en/Tutorial/FirmwareUpdater  and it will stop making the problems you mention. I have spend hours on trying to bypass the problem without a usable solution. When I found this tutorial and tried it all my problems disappeared :-)

hadwll, 2019-11-09:
I updated firmware but it did not fix.

Adding 

Client.stop()

to the end seemed to alleviate the issue.