arduino-libraries/WiFi101 #74: WiFiServer concurrent clients connected question

mhanuel26, 2016-06-25:
Hello sandeepmistry,

I have a question I would like you can clarify to me please,

Does the WiFiServer library only allow one client connection at a time? I am seeing that the WiFiServer class only has one SOCKET _socket variable even if the WINC1500 supports up to 7 TCP sockets. 

Perhaps I am just not understanding ho w_server.available can return different clients that might be available.

I will appreciate your answer,

Best

mhanuel26, 2016-06-26:
Hello sandeepmistry,

I have tested and I think either there is an issue with available() method of server or I am not getting it, when you call available method from server and a new connection arrives, the method returns the following

return WiFiClient(((flag & SOCKET_BUFFER_FLAG_SPAWN_SOCKET_MSK) >> SOCKET_BUFFER_FLAG_SPAWN_SOCKET_POS), _socket + 1);

on WiFiServer.cpp

then, I have debugged the issue, since the second connection is getting the right socket number, but reference is lost in _client array of WiFiServer class, in WiFiClient constructor a new WiFi client is added to WiFi array of pointer by doing

WiFi._client[_socket] = this;

on WiFiClient::WiFiClient(uint8_t sock, uint8_t parentsock) method from WiFiClient.cpp

But then no memory is reserve so when available returns there pointer just get lost.

I try to call available with new WiFiClient(server.available()), but same situation.

I finally get it working by returning new WiFiClass in available function of server,

This way you need to delete memory in you code, don't know if necessary but I am also doing 

	if (WiFi._client[_socket] == this){
		M2M_INFO("remove socket %d from WiFi _client pointer list", _socket);
		WiFi._client[_socket] = 0;
	}

since you assign but then when you delete memory, there is a undefined reference in client array of server.

Please let me know your thoughts,

sandeepmistry, 2016-07-11:
Hi @mhanuel26,

I've submitted #77 to try to resolve this. If you get a chance please try it, you can download it from the following link and replace `<sketch book>/libraries/WiFi101` with it. 

https://github.com/sandeepmistry/WiFi101/archive/socket-buffer.zip

mhanuel26, 2016-07-17:
Hi sandeepmistry,

I will try to integrate your changes asap and test if  there are no more issues. Will let you know if there is a problem. By the way I open I ticket at Atmel because of #70 but never got response from them, perhaps the Microchip switch has take over... but I am seeing you did some changes to socket low level library too that might fix it. 

Thank you

sandeepmistry, 2016-07-18:
Hi @mhanuel26,

Thank you for trying out the changes.

Could you please share your ticket number for #70 in a comment in that issue. I can open another support ticket to follow up on it.

mhanuel26, 2016-07-18:
Hi sandeepmistry,

Yes I will test everything, just please  wait for my feedback, I need to carefully replace your changes since I already did some changes as a workaround here. My workaround was working but I have an issue when serving a jQuery web page with other files like .css, it seems the browser open several sockets or do something that freeze the board even if I have been careful to allow access to SD card. Will try to confirm with your changes where is the problem and let you know, just give me sometime.

Thank you,

sandeepmistry, 2016-08-11:
Hi @mhanuel26, have you had a chance to test out the changes yet?

mhanuel26, 2016-08-11:
Hi sandeepmistry,

Haven't been busy with other stuff last two weeks, I will do my best to test it asap. 

Sorry for the delay.

sandeepmistry, 2016-09-16:
Hi @mhanuel26, any updates on this?

sandeepmistry, 2016-09-19:
I'm going to close this for now, I think it's a duplicate of #36.

Whenever you get a chance, please try out PR #77 and provide your feedback.

closed 2016-09-19 by sandeepmistry

KrisKasprzak, 2020-04-10:
Has anyone got multiple clients to connect? if so how?

i have a WINC1500 wifi module connected to my Teensy 3.2 and using the libraries at https://github.com/arduino-libraries/WiFi101, and I tried  this lib https://github.com/sandeepmistry/WiFi101/archive/socket-buffer.zip

After adding the line in setup(), WiFi.setPins(8, 7, 4, 2); code runs and I can connect my Android phone to this web server and point my phones Chrome browser to 196.168.1.1 and the web page generated by the server displays correctly.

However I'm trying to connect a second phone to the web server, and the phone fails to make the connection.

Anyone get this to work?