arduino-libraries/WiFi101 #75: latest firmware in wifi101 v0.9 cannot wpa personal connect to linksys router

geetee24, 2016-07-07:
help.

my new wifi 101 shield with latest firmware cannot wpa connect to linksys router running firmware 1.1.40.162464

WiFi shield present
Firmware version: 19.4.4
Firmware version required : 19.4.4

it can connect in open non encrypted to the router. 
in wpa mode, i have the correct password.

geetee24, 2016-07-08:
I tried the arduino with a new netgear router and the wifi101 can wpa connect to it just fine.

there is something wrong with the wifi101 libraary and in that it can not connect to the linksys unit.  please advise.

sandeepmistry, 2016-07-12:
Hi @geetee24,

Which model of Linksys router are you using? Do you know if the WPA mode is set to "WPA/WPA2 mixed network"?

We can open a support ticket with Atmel on your behalf, once we have more details.

geetee24, 2016-07-12:
as i already told you, yes it is in mixed mode.  I also alreadhy told 
you the router model.

sandeepmistry, 2016-07-12:
Support case 00042898 is open with Atmel to track this.

cc/ @Fede85

sandeepmistry, 2016-07-15:
We've received the following response from Atmel support:

> We tried reproducing the issue, but we did not see any issue connecting with the Linksys AP with WPA/WPA2 mixed mode security. We do not have the exact Linksys AP model, instead we used Linksys AP router model "WAP300N". If the customer still face the issue, kindly share the air capture and debug log information. 

We can provide a version of the library that enables debug logs for those interested in helping and are using a MKR1000 or Zero with the WiFi101 shield. Unfortunately, the Uno does not have enough resources to support enabling debug logging.

I will ask for more clarification on what is mean't by "air capture".

john212, 2016-08-07:
I am having exactly the same problem as gentee24. Linksys EA3500 router with latest firmware (1.1.40.162464). MKR1000 with latest firmware (19.4.4). Latest WiFi101 Library (passed firmware compatibility check for ver. 19.4.4). The MKR1000 is able to scan and find and report via the serial monitor local WiFi networks and its MAC address, type of encryption (WPA reported) and signal strength, But it will not connect to WiFi network, router is set to "WPA/WPA2 Mixed Mode Personal". The router just ignores the login attempt from the Mkr100. No errors or other problems are noted. I tried substituting the WPA passphrase key for the passphrase, no joy. Any help or advice would be appreciated. If you would like be to use the "debugging" library referred to I would be happy to but I need a little more details on how to install/use as I have not used a debugger before. Thanks for listening. -- John

geetee24, 2016-08-07:
I got no resolution from the authors of the library.  They felt that it was an issue with the router model. I disagree. 

I had to work around it by turning off the WPA WPA2 Mixed Mode personal and set it instead to just WPA2  which is a hack.  The authors of this library seem to have dropped this serious issue.

Good luck to you.

john212, 2016-08-07:
I have changed the router to WPA2 and it works. I guess now I will have to find out which legacy device around the house is still using WPA vs WPA2. It was probably time to switch to WPA2 anyway since it is more secure. Silver lining. :-) Thanks for the postings geetee24, I learned a lot by troubleshooting this, but it was also very frustrating because I couldn't find anything wrong. Cheers.

geetee24, 2016-08-07:
You are welcome.  I too spent a lot of time on the matter. I wish the authors of the library would resolve the issue.

Like you, it is time for WPA to go away but we do need the code base fixed.

sandeepmistry, 2016-08-16:
This issue can only be resolved by Atmel with a firmware update.

>  If you would like be to use the "debugging" library referred to I would be happy to but I need a little more details on how to install/use as I have not used a debugger before. 

@john212 I've pushed a branch that enables debug logging here: https://github.com/sandeepmistry/WiFi101/tree/samd-debug-enabled

You can download it as a zip and extract, then overwrite the existing `<SKETCHBOOK>/libraries/WiFi101` folder.

For the "air capture" tool Atmel referred to it's called [AirPcap](http://www.riverbed.com/products/steelcentral/steelcentral-riverbed-airpcap.html) which is quite expensive. So I doubt we can provide this capture to Atmel.

timpulver, 2016-10-20:
Same for me with library version 0.10.0 and Fritz!Box 7170 running the latest firmware.
Switching the router from WPA/WPA2 mixed mode to WPA2 solved the problem.

sandeepmistry, 2016-10-27:
@timpulver since the issue lies within the WINC1500 firmware which Atmel controls, you'll have to contact [Atmel via opening a support ticket](http://www.atmel.com/forms/Secure/verifylogin.aspx?target_url=salesforce&ssotarget=customers%2F500%2Fe%3FretURL%3D%252Fcustomers%252F500%252Fo). They have closed my original support case since I was unable to provide more details.

timpulver, 2016-10-28:
Anybody already has an account there who can open another ticket?

sandeepmistry, 2017-03-13:
There's a PR build of the IDE that contains the new v19.5.2 WINC1500 firmware: https://github.com/arduino/Arduino/pull/6069#issuecomment-286223342. Any one up for testing it with the master version of the library to see if this has been fixed?

newEndeavour, 2018-07-11:
Hi sandeepmistry, do you know if this has been fixed? I experience the same issue with my MKR1000 / WINC1500 and discovering this issue right now (WINC1500 unable to join AP network)
Thanks in advance for all the help and reference you can provide.

newEndeavour, 2018-07-15:
It turns out, v19.5.4 WINC1500 firmware fixes this issue. 
Here are the plugins:
[https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin](url)
[https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/releases/tag/v0.9.2](url)

per1234, 2022-04-02:
Closing as fixed by the firmware update.

closed 2022-04-02 by per1234