arduino-libraries/WiFi101 #76: version 0.9.1 does not work with MKR1000 and Windows Remote Arduino Experience app

keebie81, 2016-07-09:
I have an MKR1000 and in version 0.9.0 I can load the StandardFirmataWifi version 2.5.2 on the board and connect using the Windows Remote Arduino Experience app. I can also connect with the apps that I have created. But if I update WiFi101 to 0.9.1 it doesnt connect anymore in anything. On the app I get this error

connection attempt failed : a device connection was established, but the device failed handshaking procedures.
verify that your device is configured with standard firmata.
message: catastrophic failure
pin configuration not received.

sandeepmistry, 2016-07-11:
Hi @keebie81,

Do you have any instructions on how I can test this out locally? I have access to a Windows 10 VM.

I think this might be caused by the reduction of the TCP receive buffer size introduced in https://github.com/arduino-libraries/WiFi101/commit/5d6fd8779e89b0b7b15af8d8efaa3d05d7633dc5.

If you get a chance, could you please also test with pull request #77.

keebie81, 2016-07-11:
@sandeepmistry I used this app from the windows store https://www.microsoft.com/store/apps/9nblggh2041m

I tested pull #77 and it worked with the Windows Remote Arduino Experience app and my own blink app /keebie81/Blink/

sandeepmistry, 2016-09-09:
Hi @keebie81,

I just tried this with the v0.10.0 release of WiFi101, and things looked ok. I was able to toggle the LED on a MKR1000 and read the analog pins.

I think #91 might have solved the problem. I'm going to close this for now, please let us know if you still face issues and we can re-open the issue.

closed 2016-09-09 by sandeepmistry