arduino-libraries/WiFi101 #79: Please add WAP support for WiFi Access Points

davemohla12, 2016-07-19:
Currently, if you create a WiFi Access Point, you can make it open or you can use WEP security. 

Could you please add the ability to use WAP security? As you may know, WEP was deprecated in 2004 and has been replaced by WAP. WEP is significantly less secure.

Thank you!

sandeepmistry, 2016-07-19:
Hi @davemohla12,

Unfortunately the current version of the WINC1500 firmware only supports open and WEP for create an access point at this time. We would need Atmel to add support for WPA and release a new firmware version.

I suggest you "Open a support case" with Atmel Support requesting this feature and let us know of the case number here. 

http://www.atmel.com/design-support/

davemohla12, 2016-07-19:
Hi @sandeepmistry,

Sounds good. The case # is 00043231. I hope Atmel adds support for it soon.

davemohla12, 2016-07-20:
Atmel responded to the case and it looks like WAP should be supported later this year. Here's what they said:

> Access Point mode will support WPA/WPA2 securtity feature from the next ATWINC1500 firmware release, tentatively planned for September 2016.

sandeepmistry, 2017-03-13:
Closed via #148 (https://github.com/arduino-libraries/WiFi101/pull/148/commits/c4b53840c9e58cc8000691a650c44ae290f62e82).

closed 2017-03-13 by sandeepmistry