arduino-libraries/WiFi101 #82: MKR1000 wifi.status() not updating

zeesh1979, 2016-07-26:
Hi,
I made a simple app to show me value of wifi.status() at different steps but it doesnt change. i configured my MKR as access point and let my andoid mobile connect to it. it is successfully connected but when i turn off wifi, my wifi.status() doesnt show new value.

void loop() {
   Serial.println(WiFi.status());
}

above code keeps on showing me number 3 no matter my wifi is connected or not. i want to detect if wifi of device to which MKR is connected gets off

sandeepmistry, 2016-07-26:
Hi @zeesh1979,

Thanks for bringing this up. In AP mode `WiFi.status()` currently always returns `WL_CONNECTED`. I think we need to add another state to the `wl_status_t` enum. Let me think about this and come back with a proposal.

zeesh1979, 2016-07-26:
Thanks dear. I also checked in client mode where mkr1000 is connecting as client to my android phone which is acting as hotspot. I get same behavior

https://github.com/arduino-libraries/WiFi101/issues/82#issuecomment-235258220

zeesh1979, 2016-07-27:
since yesterday i was working on this and tried some way out. just now i solved my problem and thought to share with you and public.
Accidentally, I noticed that RSSI value changes with WiFi status. if wifi is connected then RSSI is around -100dbm and in connected state it is -40dbm. so i put a threshold of -60dbm for testing purpose and it solved 1st issue :)
for 2nd issue, it is solved too and i am able to detect client connectivity as well :)
below is the code. this may help you enhance the wifi101 library as well.

void loop() {
 WiFiClient client = server.available();
  if (WiFi.RSSI() >= -60){  //WiFi is connected or within range
    if (client) {           //server ready
      if (client.connected()) {   //client connected
       if (client.available()) {  //client sent data
          Serial.print("data: ");
          Serial.println(client.read());
          delay(1000);
       }
      }
    }
    else if (!client){    //server not ready
      Serial.println("waiting for client...");
      client.stop();
    }
  }
  else if (WiFi.RSSI() <= -60){   //WiFi is disconnected or out of range
    Serial.println("waiting for WiFi...");
    client.stop();
    server.begin();
  }
}

sandeepmistry, 2016-07-27:
Hi @zeesh1979,

I've submitted pull request #85 for a nicer way to use `WiFi.status()` to determine if a device is connected or disconnected to the AP. The new return values are:

* `WL_AP_LISTENING`: AP is listening, no devices connected
* `WL_AP_CONNECTED`: a device is connected to the AP

Please see the following sketch for a full example of sketch of intended usage: https://github.com/sandeepmistry/WiFi101/blob/ap-status/examples/AP_SimpleWebServer/AP_SimpleWebServer.ino

zeesh1979, 2016-07-28:
good job dude :)
i will check and share if any issues

zeesh1979, 2016-07-31:
Hi,I replaced my complete folder of wifi101 with your updated one. when i run the AP example, it says : Not used: C:\Program Files (x86)\Arduino\libraries\WiFi101exit status 1'WL_AP_LISTENING' was not declared in this scope
i checked my .cpp and .h files and definitions are ok. i thought there is conflict with another library WIFI which i deleted but still same issue (restarted IDE many times)
Date: Wed, 27 Jul 2016 09:31:58 -0700

PR #85 (2016-08-11): Add new WL_AP_LISTENING and WL_AP_CONNECTED status types for AP mode
Resolves #82.

I've added new status enums to track the connected and listening states in AP mode. As well as added some additional code to the `AP_SimpleWebServer.ino` example sketch to print out when a device connects or disconnects to/from the AP.
closed 2016-08-11 by sandeepmistry