arduino-libraries/WiFi101 #88: Add an option to switch off WiFi completely

jhnet, 2016-08-17:
I am missing the option to turn off the WiFi Chip completely after disconnecting. If you connect to a WiFi and then disconnect, the power consumption stays quite high. 

Could you please implement that or give some hints what has to be done to do so, so I can implement it on my own ?

sandeepmistry, 2016-09-09:
Hi @jhnet,

I've created pull request #100 do this, please try it out, and provide your feedback.

With it `WiFi.end()` should have the behaviour you expect.

sandeepmistry, 2016-09-16:
Closed via #100.

closed 2016-09-16 by sandeepmistry

jhnet, 2016-10-10:
Hi @sandeepmistry ,

sorry for the late reply, I had no time to test that until now. The problem is, that the power consumption does not fall after using `WiFi.end()`. I used this small test sketch:
`/*

  This example connects to an unencrypted Wifi network.
  Then it prints the  MAC address of the Wifi shield,
  the IP address obtained, and other network details.

  Circuit:
   WiFi shield attached

  created 13 July 2010
  by dlf (Metodo2 srl)
  modified 31 May 2012
  by Tom Igoe
*/
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "wifisensor";     //  your network SSID (name)
char pass[] = "2016smartportHH";  // your network password
int status = WL_IDLE_STATUS;     // the Wifi radio's status

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  pinMode(7, OUTPUT);
  digitalWrite(7, LOW);
  delay(10000);

}
int incoming = 0;
void loop() {
  if (Serial.available() > 0) {
    incoming = Serial.read();
    if (incoming == '1') {
      Serial.println("WLAN ON");
      status = WiFi.begin(ssid, pass);
      delay(10000);
      Serial.println(status);
      Serial.flush();
    }
    else if (incoming == '0') {
      Serial.println("WLAN OFF");
      WiFi.end();
      delay(10000);
      status = WiFi.status();
      Serial.println(status);
      Serial.flush();
    }
  }
}
`
After reset the MKR1000 draws about 10-12mA. When I enable the WiFi by sending a 1 it goes up to ~95-100mA. Disabling the WiFi by sending a 0 lower the power consumption a bit, but only down to ~60mA.

sandeepmistry, 2016-10-11:
@jhnet can you try the version of the code from Github master, I believe it fixes this issue.

jhnet, 2016-10-11:
Hey, I just did a quick test with the same sketch as above. Power consumption drops down to ~12mA for a short while. After a few seconds (around 10s) it rises again and stays at ~27-30mA until you switch WiFi On and Off again.

sandeepmistry, 2016-10-27:
@jhnet can you please try are removing the call to `status = WiFi.status();` in the sketch?

Calling this will actually re-initialize the WINC1500.