arduino-libraries/WiFi101 #89: TCP RST not handled? client.connected is true but client.available is false

mamama1, 2016-08-19:
Hi

using the WiFiWebServer example sketch, I have encountered a serious issue.
When hitting F5 in the browser while the page is still loading, the program gets stuck in the
`while (client.connected())` loop because client.connected stays "true" but "client.available" keeps returning "false".

Using Wireshark, I discovered that the browser (IE 11, Chrome) seems to be sending a TCP RST to reset the connection and then starts the new HTTP GET request. However, it seems that the library doesn't handle TCP RST so that we don't know that the connection should have been closed.

Maybe I'm on the wrong track, but this is all I was able to find out.

regards

sandeepmistry, 2016-09-12:
Hi @mamama1,

I've been able to reproduce the issue with WiFi101 v0.10.0 + WiFiWebServer example + Chrome on OS X and pounding CMD-R to refresh the browser.

I tried with the changes in #77 and could not reproduce. In any case, I've pushed some new changes to my [socket-buffer branch](https://github.com/sandeepmistry/WiFi101/tree/socket-buffer). Could you please download the latest and try again?

sandeepmistry, 2016-09-12:
@mamama1 could you also provide more details on what hardware you are using (MKR1000, WiFi101 shield + Uno, etc).

The testing I mentioned above was done on a MKR1000 board.

mamama1, 2016-09-15:
Hi @sandeepmistry 

Thanks for investigating!
As commented in your #77, I did try this with your socket rework initially (unsuccessfully).
However, I'll try again with the most recent version and report back as soon as I find some time.

I'm using the Adafruit WINC1500 Breakout Board + Atmega2560 (Arduino Mega). I also tried with the Uno, same results.

sandeepmistry, 2016-09-15:
Hi @mamama1,

The latest version of #77 seems ok with my WiFi101 Shield + Uno. Looking forward to hearing about your test results.

I will see if there's a smaller fix we can do in the mean time.

sandeepmistry, 2016-09-15:
A quick update, I've pushed another change to #77. Things seem better when I hold CMD-R for ~30s on OS X + Chrome.

It doesn't look we can do a smaller fix to resolve it.

mamama1, 2016-09-16:
Hi @sandeepmistry,

I've tried https://github.com/sandeepmistry/WiFi101/archive/socket-buffer.zip and I've tried https://github.com/sandeepmistry/WiFi101/archive/master.zip

Both with no luck, unfortunately. Basically the same behaviour.

This is the serial monitor output of the last two requests, before the code hangs:

> client disonnected
new client
GET / HTTP/1.1
Host: 192.168.25.161
Connection: keep-alive
Cache-Control: max-age=0
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 6.3; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
Accept-Encoding: gzip, deflate, sdch
Accept-Language: de-DE,de;q=0.8,en-US;q=0.6,en;q=0.4

client disonnected
new client
GET / HTTP/1.1
Host: 192.168.25.161
Connection: keep-alive
Cache-Control: max-age=0
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 6.3; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
Accept-Encoding: gzip, deflate, sdch
Accept-Language: client disonnected

mamama1, 2016-09-16:
i did this (Serial.println("a"); and Serial.println("c");) to know the status when it hangs, and again the serial monitor is spammed with c characters so I guess "client" is still/again connected but not "available".

``while (client.connected()) {
      Serial.println("c");
      if (client.available()) {
        Serial.println("a");
        char c = client.read();
        Serial.write(c);
....

mamama1, 2016-12-18:
Hi @sandeepmistry,

i was wondering if this will ever get resolved. is it related to the atmel firmware?

thanks

sandeepmistry, 2017-12-05:
@mamama1 if you are still interested in this, could you please try the changes in PR #204. Thanks.

sandeepmistry, 2018-01-02:
I'm closing this for now due to lack of feedback, please re-open if you are still interested and have tried the changes for PR #204 which have been merged.

closed 2018-01-02 by sandeepmistry

i486you, 2018-11-06:
The problem is that some web browsers request for favicon.ico. Chrome requests it every reload but  Firefox, Safari and IE don't. 

And the sample code does not check the second request for favicon.ico.  So I edited some of them.

This question is posted in 2016. However, the sample code and the library [ESP8266Wifi ](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi) have not fixed. So I hope this code helps someone.

```
void WaitWebClient()
{  
     WiFiClient client = server.available();

     // wait for a client (web browser) to connect
     if (client)
     {
          Serial.println("\n[Client connected]");

          // <<<<<<<<< Added
          // In chrome browser, if you close the tab or window opened during a request
          // client.connected() will be true and client.available() will be zero.
          int cntToBreak = 0;

          while (client.connected())
          {
               // available() returns the size of message.
               if(client.available())
               {
                    String line = client.readStringUntil('\r');
                    client.readStringUntil('\n');        // <<<<< Added
                    Serial.print(line);
        
                    // wait for end of client's request, that is marked with an empty line
                    // <<<<<<<<<<< Modified
                    if(line.length() == 0)
                    {
                         client.println(prepareHtmlPage());
                         break;
                    }

                    Serial.println();
                    Serial.println(line.length());
               }

               // <<<<<<<<<<<<< Added
               // Client is connected but no message.
               // then count the times of this loop.
               else
               {
                    if(cntToBreak++ > 20)
                    {
                         cntToBreak = 0;
                         break;
                    }
               }
          }
         delay(1); // give the web browser time to receive the data

          // close the connection:
          if(client.stop())
          {
               Serial.println("[Client disonnected]");
          }
     }
}
```