arduino-libraries/WiFi101 #90: Atmel Update firmware 19.5.x

trlafleur, 2016-08-22:
In talking with my local FAE for Microchip (Atmel) I was advised that the new chip firmware 19.5.x is in beta testing with an expected release data of late Oct 2016.

trlafleur, 2016-10-25:
Microchip (atmel) has now delayed firmware update on this device until mid Jan 2017.

trlafleur, 2017-01-27:
Code is in RC6. Release date of late February 2017

sandeepmistry, 2017-03-06:
I've opened the following PR to add support for the new firmware: https://github.com/arduino-libraries/WiFi101/pull/148

It needs lot of testing before it can be merged and released.

sandeepmistry, 2017-03-13:
Closed via #148. We'll work on the new firmware being available in the hourly IDE builds next.

closed 2017-03-13 by sandeepmistry

sandeepmistry, 2017-03-13:
There's a PR build that contains the new firmware updater and the IDE available here for testing: https://github.com/arduino/Arduino/pull/6069#issuecomment-286223342.