arduino-libraries/WiFi101 #92: [Feature request] Add compatibility with "Wifi direct"

q2dg, 2016-08-25:
ATWINC1500 module that support Wifi Direct/P2P according to the datasheet. Just wondering if it is feasible.
Thanks!

sandeepmistry, 2016-09-06:
Hi @q2dg,

We can certainly consider integrating the changes from the fork.

Do you have any ideas on your use cases for P2P mode? I tried it out, but it only seems to work with Android, so it might be confusing to the user base.

From the ["WINC1500 supported connection modes and limitations" page](http://atmel.force.com/support/articles/en_US/FAQ/WINC1500-supported-connection-modes-and-limitations):

> WiFi Direct or P2P mode.
> * STA and P2P concurrency is not supported. 
> * Device can only operate as a WiFi direct device not as a host.

So you can't get 2 WINC1500 based boards to use P2P mode again.

q2dg, 2016-09-06:
Ok, let's leave it be, then. With the recent addition of AP-mode with beginAP() API,  this feature it isn't so urgent, I think...so I'll close the issue. Free open it if you consider it. Thanks a lot for your job!!

closed 2016-09-06 by q2dg

sandeepmistry, 2016-09-06:
Thanks @q2dg! I agree.

We can re-open once more devices support P2P mode or there is some demand + use cases for the feature. For anyone else who comes across this thread, please share you're thoughts on the feature :)