arduino-libraries/WiFi101 #93: HTTP provisioning does not connect

apruikko, 2016-08-30:
I was having problems with provisioning not connecting to Access Point after entering credentials in WINC1500 provisioning web page.

Issue description [http://forum.arduino.cc/index.php?topic=420962.new#new]

The example codes in Atmel Studio Framework had a function call m2m_wifi_set_device_name() before m2m_wifi_start_provision_mode(). When I added m2m_wifi_set_device_name() call to WiFi.cpp function  WiFiClass::beginProvision(char *ssid, char *url, uint8_t channel) the connection to provisioned AP was successfull.

[provision_http_example.zip](https://github.com/arduino-libraries/WiFi101/files/444525/provision_http_example.zip)

sandeepmistry, 2016-09-06:
Hi @apruikko,

I've just tried this with the master version of the library on a MKR1000 and Zero + WiFi 101 shield. I didn't have any issue.

Here's my test sketch:

```arduino
#include <SPI.h>
#include <WiFi101.h>

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  Serial.println("Starting to provision mode");
  // change the next line to use a different channel
  // channels 1 (default), 6, and 11 are supported
  while (WiFi.beginProvision("blah", "wifi101.arduino.cc") != WL_CONNECTED) {
    Serial.print(".");
    delay(5000);
  }
  Serial.println();
  Serial.println("Connected!");

  printCurrentNet();
  printWifiData();
}

void loop() {
  // check the network connection once every 10 seconds:
  delay(10000);
  printCurrentNet();
  printWifiData();
}

void printWifiData() {
  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
  Serial.println(ip);

  // print your MAC address:
  byte mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC address: ");
  Serial.print(mac[5], HEX);
  Serial.print(":");
  Serial.print(mac[4], HEX);
  Serial.print(":");
  Serial.print(mac[3], HEX);
  Serial.print(":");
  Serial.print(mac[2], HEX);
  Serial.print(":");
  Serial.print(mac[1], HEX);
  Serial.print(":");
  Serial.println(mac[0], HEX);
}

void printCurrentNet() {
  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.println(rssi);

  // print the encryption type:
  byte encryption = WiFi.encryptionType();
  Serial.print("Encryption Type:");
  Serial.println(encryption, HEX);
  Serial.println();
}
```

There is no need to call `m2m_wifi_set_device_name` from my tests. Could you please share the sketch you were testing with?

I'm not sure how useful the `WiFi.beginProvision` functionality is, since you will be required to re-provision every time the board is power cycled.

sandeepmistry, 2016-09-19:
I'm closing this for now due to lack of feedback and since the feature is working for me.

Feel free to re-open with requested info. if you are still interested.

closed 2016-09-19 by sandeepmistry