arduino-libraries/WiFi101 #99: Change IP Address of an Access Point from .config()

3nt0, 2016-09-09:
I think it will be useful if the address of an Access Point could be changed by the `WiFi.config();` command, instead of modifying the WiFi.cpp library.
Thanks.

sandeepmistry, 2016-09-09:
Hi @3nt0,

Good suggestion! I've create pull request #101 for this change. Please try it out and provide your feedback.

3nt0, 2016-09-11:
Ok, I've tried with 2 client (my phone).
I can change the IP of the AP, with no problem.
But the IP address of the client connected always start from x.x.x.100 and if someone set the Access Point's IP at something like 192.168.90.100 it doesn't work.

sandeepmistry, 2016-09-12:
Hi @3nt0,

Thank you for trying out the pull request!

> But the IP address of the client connected always start from x.x.x.100 and if someone set the Access Point's IP at something like 192.168.90.100 it doesn't work.

I think this is a limitation of the WINC1500 firmware (which Atmel controls). It looks like the DHCP server code is hard coded to use `x.x.x.100` as the client IP.

We can make `WiFi.beginAP(...)` fail if `WiFi.config(...)` was used with an IP address that ends with `.100`. What do you think?

Just curious, why are you interested in changing the IP address of the board in AP mode?

3nt0, 2016-09-12:
Hi sandeepmistry,

> I think this is a limitation of the WINC1500 firmware (which Atmel controls). It looks like the DHCP server code is hard coded to use x.x.x.100 as the client IP.

In my opinion, it would be great if Atmel give us the possibility to change that, leaving that hard coded it's a strange limitation (or may be): from a wifi device I expect these functionality.
Have you got any experience with requests to Atmel? Do they "listen"?

> We can make WiFi.beginAP(...) fail if WiFi.config(...) was used with an IP address that ends with .100. What do you think?

That could be a solution, but I think that type of limitation must be well documented.

> Just curious, why are you interested in changing the IP address of the board in AP mode?

I'm replacing a PLC and a wifi AP that dialogue to an android app, with MKR1000 and a few board for 4..20mA analog input sensor where my net address are 192.168.90.150 for the AP and 192.168.90.151 for the android phone.
Yes, my boss want a cheaper solution than PLC...
And yes, it works also with the 192.168.1.x net but force me to change standard.
The last... if WiFi.config() change the IP of the connection to an Wifi, why shouldn't change the IP of an AP? =)

Thanks

sandeepmistry, 2016-09-13:
> Have you got any experience with requests to Atmel? Do they "listen"?

Yes, we have some success. I would suggest you open a support case with them and include a link to this issue: http://www.atmel.com/design-support/

Please also add your case number as a comment here.

>> We can make WiFi.beginAP(...) fail if WiFi.config(...) was used with an IP address that ends with .100. What do you think?
>
> That could be a solution, but I think that type of limitation must be well documented

Ok, let's go for this in the short term. I will add this to PR #101 shortly.

> I'm replacing a PLC and a wifi AP that dialogue to an android app, with MKR1000 and a few board for 4..20mA analog input sensor where my net address are 192.168.90.150 for the AP and 192.168.90.151 for the android phone.

Does the Android side use a static IP or DHCP?

sandeepmistry, 2016-09-13:
Added https://github.com/sandeepmistry/WiFi101/commit/ec7bd3f19635ad530050e3845cb4e6deac2d4c7d to #101.

sandeepmistry, 2016-09-16:
Closed via #101.

closed 2016-09-16 by sandeepmistry