arduino-libraries/WiFi101 #104: Unnecessary 10 second delay

ladyada, 2016-09-21:
hiya, all the examples have
```
    // wait 10 seconds for connection:
    delay(10000);
```
but this is rarely necessary and makes the WINC1500 seem slower than it really is. usually connections are 2 seconds or less

suggest replacing with:

```
    // wait up to 10 seconds for connection:
    uint8_t timeout = 10;
    while (timeout && (WiFi.status() != WL_CONNECTED)) {
      timeout--;
      delay(1000);
    }
```

sandeepmistry, 2016-09-26:
Hi @ladyada,

I agree the delay is necessary, the 2 seconds you mention after calling `WiFi.begin(...)` can also be removed, by bumping the 60s timeout in [`WiFiClass::startConnect `](https://github.com/arduino-libraries/WiFi101/blob/master/src/WiFi.cpp#L363-L368) to 75s or 90s, this is really the timeout for waiting for a DHCP lease after connect.

Would you mind trying that out?

ladyada, 2016-09-26:
hiya! are you saying you think that startConnect() should poll WiFi.status() for up to ~3-5 seconds or...?

sandeepmistry, 2016-09-26:
It's already doing so:

```c++
	// Wait for connection or timeout:
	unsigned long start = millis();
	while (!(_status & WL_CONNECTED) &&
			!(_status & WL_DISCONNECTED) &&
			millis() - start < 60000) {
		m2m_wifi_handle_events(NULL);
	}
	if (!(_status & WL_CONNECTED)) {
		_mode = WL_RESET_MODE;
	}
```

However, I think in your case the AP association +  DHCP might be taking longer than 60s (which is odd). There are some commented out `SERIAL_PORT_MONITOR.print` statements in WiFi101.cpp `wifi_cb`, if you enable those we should get a better sense on the timing of AP association and DHCP timing. I would also suggest printing out the value of `millis()` with them.

mjlitke, 2018-09-28:
I've had similar thoughts regarding the examples. I also wondered why the loops keep calling WiFi.begin(ssid, pass), versus calling it once and checking WiFi.status(). I looked at the code in WiFi.cpp, and it appears the main difference between the two is that begin() calls m2m_wifi_connect(). Is it necessary to keep calling m2m_wifi_connect() for each iteration of the loop. I have been using:

```
   WiFi.begin (ssid, pass);
   end_ms = millis() + (<timeout in seconds> * 1000);
   while (millis() < end_ms){
      if (WL_CONNECTED == WiFi.status()){
         break;
      }
      delay(200);
   }
```

and so far haven't noticed an issues, but my testing has been somewhat limited.

Another question I had (and please let me know if it enough of a different topic that I should move it somewhere else) is if I update the library to include allowing the timeout to be set when waiting for a connection (which is shown in the WiFiClass::startConnect() code snippet included in the previous message), is that something you would be open to incorporating? If so, my initial thought is to add a private _timeout variable to the WiFi class, and add a WiFi.setTimeout (<timeout>) method (similar to the Steam class functionality). If the method is not called it would default to 60000 ms, which is the current behavior. Looking at the WiFi.cpp code, a hard coded value of 60000 ms appears in WiFi.begin() and WiFi.startConnect (), and would get replace with the _timeout variable.

If you are open to adding this, please let me know, and I will create a pull request with the updates.