arduino-libraries/WiFi101 #107: AP & WEP doesn't work for all devices connected

3nt0, 2016-10-04:
When the configuration of the MKR1000 is on Access Point with WEP security some devices connect and get the IP from DHCP, other devices not.
If I change the AP with no security every devices connect and get IP.

The password is 10 character long.

How can I debug that? Where do the system hangs?

sandeepmistry, 2016-10-04:
Hi @3nt0,

Can you please share the sketch you are using the MKR1000? Also, have you tried changing the WEP key index?

Which devices are having issues?

3nt0, 2016-10-04:
No, I only tried with WEP and no security wifi.
The best would be WPA but I've read that's not working.

Works with WEP?
YES - OPO with paranoid android
NO - OPO with CyanogenOS
NO - One Plus Three CyanogenOS
YES - Motorola Moto G 1st gen with motorola ROM
NO - MacMini 2012
NO - HP laptop with Win7

My phone is the first OPO so I never noticed.

Here it is:

```
#include <SPI.h>
#include <WiFi101.h>
#include <Wire.h>
#include <MCP342x.h>

// ----------------------------------------------------------

int led =  LED_BUILTIN;
char ssid[] = "EJ775EP"; // created AP name
char pass[] = "1234567890";      // AP password (needed only for WEP, must be exactly 10 or 26 characters in length)
int keyIndex = M2M_WIFI_SEC_WEP;                // your network key Index number (needed only for WEP)
int status = WL_IDLE_STATUS;
WiFiServer server(80);
bool nocltoAP;
// the IP address for the shield: (solo per collegamenti ad access point esistenti)
IPAddress ip(192, 168, 90, 150);
//IPAddress dns(8, 8, 8, 8);
//IPAddress gw(0, 0, 0, 0);
//IPAddress mask(255, 255, 255, 0);
char buffer[50];

// ----------------------------------------------------------

uint8_t address = 0x6D;
MCP342x adc = MCP342x(address);

// ----------------------------------------------------------

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.println("Access Point Web Server");

  pinMode(led, OUTPUT);      // set the LED pin mode

  // check for the presence of the shield:
  if (WiFi.status() == WL_CONNECT_FAILED) {
    Serial.println("WiFi shield not present");
    // don't continue
    while (true);
  }

  WiFi.config(ip);

  // print the network name (SSID);
  Serial.print("Creating access point named: ");
  Serial.println(ssid);
  // Create open network. Change this line if you want to create an WEP network:
  if (WiFi.beginAP(ssid, keyIndex, pass, 4) != WL_AP_LISTENING) {
    Serial.println("Creating access point failed");
[... 137 lines cut ...]
  }
  if (WiFi.status() ==  WL_AP_FAILED){
    Serial.println("WL_AP_FAILED");
  }

  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rss);
  Serial.println(" dBm");
  // print where to go in a browser:
  Serial.print("To see this page in action, open a browser to http://");
  Serial.println(ip);
}

```

sandeepmistry, 2016-10-07:
@3nt0 thanks for the details. Atmel needs to provide a firmware update to support WPA in AP mode(see #79), so it is not something we change in this library for now.

Have you tried changing the key index or channel?

```
WiFi.beginAP(ssid, keyIndex, pass, channel);
```

`keyIndex` can be `1` to `4`, channel `1` to `12`, the default channel is `1` if not provided.

If that doesn't make a difference, I think it's a firmware issue so we would have to report it to Ateml. Let me know.

3nt0, 2016-10-08:
Hi @sandeepmistry 
I think the error is this one:
`int keyIndex = M2M_WIFI_SEC_WEP;`
I don't know why but I've put that "M2M_WIFI_SEC_WEP", a mistake for sure. =(

Now it seems working on, but I'll continue testing until I have a good number of devices... and that takes me a week or two.

sandeepmistry, 2016-10-11:
@3nt0 thanks for the update. I'll close this for now, we can re-open if you still face issues with other devices.

closed 2016-10-11 by sandeepmistry