arduino-libraries/WiFi101 #108: Add auto-reconnect support

sandeepmistry, 2016-10-07:
In order to improve sketch reliability, we should look into attempting to re-connect to AP's when a disconnect event occurs. This would remove the burden from the user's sketch.

drp0, 2016-10-10:
This code tries 3 times to connect.
The minimum wait for a connection is 2 seconds.
Within void loop, the connection is checked every 5 minutes.
If the 3 connection attempts fail in the loop, there is a 20second delay before trying again.
 
```C++
/*
3 tries at connecting
No 10 second wait
Retest connection after checkstatus milliseconds in loop
Re-attempt connection every 20 seconds if 1st attempt(s) fail

Wifi status codes
  WL_NO_SSID_AVAIL    (1) : assigned when no SSID are available;
  WL_SCAN_COMPLETED   (2) : assigned when the scan networks is completed;
  WL_CONNECTED        (3) : assigned when connected to a WiFi network;
  WL_CONNECT_FAILED   (4) : assigned when the connection fails for all the attempts;
  WL_CONNECTION_LOST  (5) : assigned when the connection is lost;
  WL_DISCONNECTED     (6) : assigned when disconnected from a network;
  WL_AP_LISTENING     (7)
  WL_AP_CONNECTED     (8)
  WL_AP_FAILED        (9)
*/

#include <SPI.h>

#include <SdFat.h>
SdFat sd; // file system object
Sd2Card card;

#include <WiFi101.h>

int port = 8082;
WiFiServer server(port);

const byte wifiSelect = 10;
const byte sdSelect = 53;
const byte attempts = 3; // number of tries at connection
const unsigned long checkstatus = 300000; // check status every 5 minutes

unsigned long oldTime;
char ssid[] = "DRP3";    // network name whatever 

IPAddress ip; // ip address of wifi101
int quality; // connection quality
int wifistatus = WL_IDLE_STATUS;    // the Wifi radio's status

boolean report = true; // Serial output?

void die(){ // no functioning wifi 101 card
  if  (report) Serial.print(F("Error: No WIFI card"));
// any warning process  
  while(1);
}

void setup() {
  if(report){
  Serial.begin(115200);
    while (!Serial) yield(); 
  }
pinMode(wifiSelect, OUTPUT);
pinMode(sdSelect, OUTPUT);

// nb1 issue sd.begin(sdSelect, SPI_FULL_SPEED)) before wifi
  // or else the card is corrupted!
  if (!sd.begin(sdSelect, SPI_FULL_SPEED)) {
    if (report) Serial.println(F("sd.begin failed"));
    while(1);
  }
  if (!card.init(SPI_FULL_SPEED, sdSelect)) {
    if (report) Serial.println(F("card.init failed!"));
    while(1);
  }
// nb2 if sdSelect != default select
  //need to make defalut select output

  if (wifistatus == WL_NO_SHIELD) die();
  
  if (!ssidConnect(ssid, attempts) ) {
    if (report) Serial.println(F("Abort"));
    
    while(1){ // connection has failed
    // error message, bang a drum, etc
    delay(3000);
    }
  }
oldTime = millis();
}

void loop() {
unsigned long timeNow = millis();
  if (timeNow - oldTime >= checkstatus){ // check wifi every 5 minutes (checkstatus = 300000)
  oldTime = timeNow; // reset timeout
  wifistatus = WiFi.status();
    if(wifistatus == WL_NO_SHIELD) die();
     
    if( (wifistatus == WL_CONNECTED) || (wifistatus > WL_DISCONNECTED) ) {
    quality = WiFi.RSSI();
    }else{ // reconnect if not AP_listening, AP_connected , AP_failed or dead!
      while (WiFi.status() != WL_CONNECTED){
        if(report) Serial.println(F("WiFi down"));
        
        if (ssidConnect(ssid, attempts)){ // try to re-connect
        oldTime = millis();
        break;
        }else{
          if(report) Serial.println(F("Re-trying in 20 seconds"));
        delay(20000); // 20 seconds to trying again
        }
      }
    }
    
    if(report) {
    Serial.print(F("WiFi on, quality "));
    Serial.print(quality); Serial.println(F(" dBm"));
    }  
  }
// rest of loop stuff
}

boolean ssidConnect(String thessid, byte attempts){
byte mytry = 0;
  // attempt to connect to Wifi network:
  do { // try attempts times
  mytry ++;
    if (report) {
    Serial.println(F("Connect attempt ")); Serial.print(mytry);
    Serial.print(F(" to SSID: ")); Serial.println(thessid);
    }
  // Connect to network.
  wifistatus = WiFi.begin(thessid);
  // wait up to 10 seconds for connection:
  unsigned long starttime = millis();
    do { // check status every 2 seconds
    delay(2000);
    wifistatus = WiFi.status();
    } while (wifistatus != WL_CONNECTED && millis() - starttime < 10000); 
  } while (wifistatus != WL_CONNECTED && mytry < attempts); // 3 attempts

  if (wifistatus != WL_CONNECTED){
    if (report) {
    Serial.print(F("Unable to connect to ")); Serial.println(thessid);
    }
  return false;
  }
  if (report) {
  Serial.print(F("Connected, commencing server on port "));
  Serial.println(port);
  }

server.begin(); // if required
ip = WiFi.localIP();
quality = WiFi.RSSI();
  if (report) printinfo();
return true;
}
void printinfo(){
Serial.println(F("\n********************************"));
// print your WiFi shield's IP address:
Serial.print(F("IP Address: "));
Serial.println(ip);
// print your MAC address:
byte mac[6];
WiFi.macAddress(mac);
Serial.print(F("MAC address: "));
  for(byte i = 5; i > 0; i--) {
  Serial.print(mac[i], HEX);
  Serial.print(":");
  }
Serial.println(mac[0], HEX);
// print your subnet mask:
IPAddress subnet = WiFi.subnetMask();
Serial.print(F("NetMask: "));
Serial.println(subnet);
// print your gateway address:
IPAddress gateway = WiFi.gatewayIP();
Serial.print(F("Gateway: "));
Serial.println(gateway);
// print the received signal strength:
Serial.print(F("signal strength (RSSI):"));
Serial.print(quality); Serial.println(F(" dBm"));
// print the encryption type:
Serial.print(F("Encryption Type: "));
printEncryptionType(WiFi.encryptionType() );
Serial.println(F("*******************************\n"));
Serial.flush(); 
}

void printEncryptionType(int thisType) {
  // read the encryption type and print out the name:
  switch (thisType) {
    case ENC_TYPE_WEP:
      Serial.println(F("WEP"));
      break;
    case ENC_TYPE_TKIP:
      Serial.println(F("WPA"));
      break;
    case ENC_TYPE_CCMP:
      Serial.println(F("WPA2"));
      break;
    case ENC_TYPE_NONE:
      Serial.println(F("None"));
      break;
    case ENC_TYPE_AUTO:
      Serial.println(F("Auto"));
      break;
  }
}
``'

sandeepmistry, 2016-10-11:
Hi @drp0,

Thanks for sharing! I was thinking this could be integrated into the WINC1500 callback somehow.

ericbaril72, 2016-12-14:
+1 to integrate auto-reconnect.
How about a flag that could activate AP_Mode when Wifi not-avail/ no connect /disconnects ?

I've had a few difficulties when my devices would not connect to the router(s)
--> Need to recompile 

Having AP_Mode as a fallback could let us change or add new ssid/pass combinations.
I would have code to share really soon on this combo ... unfortunatly, my test setup identifies many other "glitches" and I still cannot clearly see if it's Wifi.AP/Wifi.begin or my server connections.

let me know your thoughts, i could easily "taylor" my current code based on feedback

sandeepmistry, 2016-12-14:
@ericbaril72 we're exploring leveraging the WINC1500's provisioning mode in PR #102.

It would be great if you could try it out, and provide feedback on the changes in the pull request.

youcangetme, 2017-03-19:
This has become so annoying! This is fast becoming one of my biggest grumpy points.

I have code that allows the unit to slip into AP mode if no access point is available for which it has keys for. You then can join its network and use a Web page to scan and join another network.

Now I have the issue where for whatever reason the unit drops the connection and never rejoins. It would seem like the library would say, "oh, hey! I'm no longer connected! Let's reconnect!"

I do not seem to be able to reconnect all of the servers I created after I try something like the above code that @drp0 offered. They respond via code access but neither inbound or outbound traffic seems to be recognize.

drp0, 2017-03-20:
I had a simillar issue and solved it by issuing:
WiFi.end();
when the wifi connection loss was first detected and prior to the first re-connection attempt.
David