arduino-libraries/WiFi101 #110: AP Mode.  Unable to re-connect

driverblock, 2016-10-17:
Sample code attached.
Code creates an AP, then starts an HTTP server.  Server works normally.  If you disconnect computer from AP and then reconnect, Browser is no longer able to connect to server.
Adafruit forum thread here:
https://forums.adafruit.com/viewtopic.php?f=57&t=104876&p=525448#p525215

(also, statement  if (WiFi.beginAP(ssid) != WL_CONNECTED) { always fails, but AP is still created. What is correct test?)

`#include <WiFi101.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

#include <SPI.h>
//#include <Adafruit_WINC1500.h>

// Define the WINC1500 board connections below.

#define WINC_CS   8
#define WINC_IRQ  7
#define WINC_RST  4
#define WINC_EN   2     // or, tie EN to VCC

// Setup the WINC1500 connection with the pins above and the default hardware SPI.
//Adafruit_WINC1500 WiFi(WINC_CS, WINC_IRQ, WINC_RST);

// **** WARNING ****
// If running this example on an Uno or compatible you MUST change the LED_PIN
// value below from 13 to something else that doesn't conflict with the SPI
// pins.  Try pin 9 for example.
// **** WARNING ****
#define LED_PIN  13  // This example assumes you have a LED connected to pin 13
// (with a resistor to limit current!).  Connect LED anode
// (longer leg) to pin 9, then LED cathode (shorter pin)
// through a resistor (~300-1k ohm) to ground.

#define a_button 9
#define b_button 6
#define c_button 5

char ssid[] = "Load Bank";      //  created AP name
char pass[] = "besl";         // (not supported yet)

float volts = 0;
float amps = 0;
float bhp = 0;
int voltsint = 0;
int ampsint = 0;
int bhpint = 0;

int status = WL_IDLE_STATUS;
//Adafruit_WINC1500Server server(80);
WiFiServer server(80);
IPAddress ip;

// *****  Sub Routines  *****

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  //IPAddress ip = WiFi.localIP();
  ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
[... 169 lines cut ...]

          break;
        }
        if (c == '\n') {
          // you're starting a new line
          currentLineIsBlank = true;

        }
        else if (c != '\r') {    // if you got anything else but a carriage return character,
          //currentLine += c;      // add it to the end of the currentLine

          // you've gotten a character on the current line
          currentLineIsBlank = false;
        }
      }
    }
    // give the web browser time to receive the data
    delay(1);

    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
`

sandeepmistry, 2016-10-27:
Hi @driverblock,

Thanks for reporting this, I'm closing it as a duplicate of https://github.com/arduino-libraries/WiFi101/issues/46.

Right now `server.begin()` needs to be called again everytime a device disconnects from the AP. The changes proposed in #77 might fix this, but they need more testing.

closed 2016-10-27 by sandeepmistry

driverblock, 2016-11-01:
Thanks for looking into it!  I've passed the status along to the adafruit forum user.