arduino-libraries/WiFi101 #111: V10 Library fails to compile

drp0, 2016-10-17:
With github source
[https://github.com/arduino-libraries/WiFi101/tree/9384c5b7b3c39d756877f728fbc346b38871ee77](url)

I get the following compilation errors:
Arduino: 1.6.11 (Windows XP), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Documents and Settings\Gabbitus\My Documents\Arduino\libraries\WiFi101\src\WiFi.cpp: In member function 'uint32_t WiFiClass::getTime()':

C:\Documents and Settings\Gabbitus\My Documents\Arduino\libraries\WiFi101\src\WiFi.cpp:922:13: error: aggregate 'WiFiClass::getTime()::tm tm' has incomplete type and cannot be defined

   struct tm tm;

             ^

C:\Documents and Settings\Gabbitus\My Documents\Arduino\libraries\WiFi101\src\WiFi.cpp:932:17: error: 'mktime' was not declared in this scope

   t = mktime(&tm);

                 ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

David

sandeepmistry, 2016-10-27:
Hi @drp0,

There was a change in https://github.com/arduino-libraries/WiFi101/pull/84 that required `time.h` and a never version of the AVR core. I've opened PR #113 to make the Github master version of the library compatible with older AVR cores. Please try out the changes and provide your feedback.

drp0, 2016-10-28:
I am using Arduino ide 1.6.12
I have time by michael margolis version 1.50 (latest)
With wifi101 version 0.10.0.
On an arduino mega 2560

The library still does not compile if the sketch includes the <TimeLib.h> (time) library:
I had to force a rebuild between toggling the inclusion of TimeLib.h

The errors returned are 
Arduino: 1.6.12 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
C:\Users\laptop\Documents\Arduino\libraries\YiFi101\src\WiFi.cpp: In member function 'uint32_t WiFiClass::getTime()':

C:\Users\laptop\Documents\Arduino\libraries\YiFi101\src\WiFi.cpp:929:13: error: aggregate 'WiFiClass::getTime()::tm tm' has incomplete type and cannot be defined

   struct tm tm;

             ^

C:\Users\laptop\Documents\Arduino\libraries\YiFi101\src\WiFi.cpp:939:17: error: 'mktime' was not declared in this scope

   t = mktime(&tm);

                 ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

The sketch is from the new ping example
```C++
#include <SPI.h>
#include <WiFi101.h>

#include <TimeLib.h>

char ssid[] = "DRP3";     //  your network SSID (name)
int status = WL_IDLE_STATUS;     // the Wifi radio's status

// Specify IP address or hostname
String hostName = "www.google.com";
int pingResult;
int loopcount = 0;

void setup() {
  // Initialize serial and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network:
    status =  WiFi.begin(ssid);

    // wait 5 seconds for connection:
    delay(5000);
  }

  // you're connected now, so print out the data:
  Serial.println("You're connected to the network");
  printCurrentNet();
  printWifiData();
}

void loop() {
loopcount++;
  if(loopcount == 10) while(1);
  Serial.print("Pinging ");
  Serial.print(hostName);
  Serial.print(": ");

  pingResult = WiFi.ping(hostName);

  if (pingResult >= 0) {
    Serial.print("SUCCESS! RTT = ");
    Serial.print(pingResult);
    Serial.println(" ms");
  } else {
    Serial.print("FAILED! Error code: ");
    Serial.println(pingResult);
  }

  delay(5000);
}

void printWifiData() {
  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP address : ");
  Serial.println(ip);

  Serial.print("Subnet mask: ");
  Serial.println((IPAddress)WiFi.subnetMask());

  Serial.print("Gateway IP : ");
  Serial.println((IPAddress)WiFi.gatewayIP());

  // print your MAC address:
  byte mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC address: ");
  Serial.print(mac[5], HEX);
  Serial.print(":");
  Serial.print(mac[4], HEX);
  Serial.print(":");
  Serial.print(mac[3], HEX);
  Serial.print(":");
  Serial.print(mac[2], HEX);
  Serial.print(":");
  Serial.print(mac[1], HEX);
  Serial.print(":");
  Serial.println(mac[0], HEX);
  Serial.println();
}

void printCurrentNet() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print the MAC address of the router you're attached to:
  byte bssid[6];
  WiFi.BSSID(bssid);
  Serial.print("BSSID: ");
  Serial.print(bssid[5], HEX);
  Serial.print(":");
  Serial.print(bssid[4], HEX);
  Serial.print(":");
  Serial.print(bssid[3], HEX);
  Serial.print(":");
  Serial.print(bssid[2], HEX);
  Serial.print(":");
  Serial.print(bssid[1], HEX);
  Serial.print(":");
  Serial.println(bssid[0], HEX);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI): ");
  Serial.println(rssi);

  // print the encryption type:
  byte encryption = WiFi.encryptionType();
  Serial.print("Encryption Type: ");
  Serial.println(encryption, HEX);
  Serial.println();
}
```

It may be worth adding to any ping documentation that a dmz might be required on the yifi101 lan ip, to get the ping to work. 

David

sandeepmistry, 2016-10-28:
@drp0 

> I have time by michael margolis version 1.50 (latest)

This should not be needed, because the new AVR compiler includes `time.h`.

Can you remove the `#include <TimeLib.h>` from the sketch and try again. Did you try my changes in #113 yet?

Also, please share what `Tools -> Board ... -> Board Manager` shows for the "Arduino AVR Boards" version.

drp0, 2016-10-29:
Hi Sandeep,
Arduino AVR Boards version 1.6.14 installed.
The timeLb is required for my purposes and is  the library advised in the arduino payground.
I am using the time library now() and derivative functions from Timelib.h
These do not work with the included time.h library
I followed the link advised in [http://playground.arduino.cc/Code/Time](url)
The time functions are used to set the arduino time following a yifi101 call to a ntp server-
I suggest that I will not be the only one to want to do this!

Re removing the time library. I believed I covered that in the post above ("toggling")
When not present, the ping sketch does compile.

David

trlafleur, 2016-10-29:
I had the same issue....

sandeepmistry, 2016-10-31:
Please note the playground section of the website is user editable, the content is not created by Arduino, but instead by the community.

I was able to reproduce with Paul Stoffregen's Time library: https://github.com/PaulStoffregen/Time - I think it's the same one you are using. The problem is (at least on my Mac), that `Time.h` from Paul's library is included instead of the `time.h` provided by AVR GCC. If I rename `Time.h` to `Time_.h` things build successfully.

sandeepmistry, 2016-10-31:
@cmaglie @facchinm any suggestions on how to proceed with this?

It seems like the `Time.h` header from the Time library overrides the `time.h` header included by AVR GCC. I would have expected the AVR GCC header to take priority.

drp0, 2016-10-31:
Hi Sandeep,
Re. > I was able to reproduce with Paul Stoffregen's Time library
This is the correct time library.
I discovered it by linking from https://www.arduino.cc/en/Reference/Libraries
I appreciate your point regarding playground editing.
However, anyone looking for a time library will start on the same page as me and end up with Paul's library.
I have been unable to find any reference to a core time library.
Therefore I do not know what functions it supports and therefore whether it could be used in-stead of Paul's library.
now(), hour(0, minute(), second(), day() etc are not natively supported.
Do you know of a url supplying details of the core time library?
If so, I am happy to take a look at it.
If the mac's are using Paul's library, why not bundle the same for all platforms?

David

sandeepmistry, 2016-11-02:
> I have been unable to find any reference to a core time library.

It's the one provided by the compiler `Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include/time.h`. More details here: http://www.nongnu.org/avr-libc/user-manual/group__avr__time.html

> If the mac's are using Paul's library, why not bundle the same for all platforms?

We don't really want to include `Time.h` from Paul's lib, because it doesn't declare `struct tm` or `mktime`.

The problem is because of the case insensitive file system in OS X and Windows XP, `Time.h` from Paul's library is overriding the `<time.h` include.

I've pushed a change to #113, to check if the system `<time.h>` is included or overridden. If it is not, like when someone includes `#include <TimeLib.h>`, then `WiFi.getTime()` will return 0, because we are not sure if `struct tm` or `mktime` is present. I think this is an ok compromise.

@drp0 @trlafleur please try out the latest change from PR #113.

PR #113 (2016-11-14): Backwards compatibility for older AVR core versions
They do not have time.h. Resolves #111.

cc/ @drp0.
closed 2016-11-14 by agdl