arduino-libraries/WiFi101 #112: set device WiFi hostname

hanabanana, 2016-10-25:
It seems there is no API to set the device hostname and only the IP address can be queried using WiFi.localIP(). With the IP address possibly changing with a DHCP server, it seems difficult to get the device address and/or map the device with its corresponding IP given multiple boards present. The mDNS implementation seems to be an option but it requires software like Bonjour on different OS. 

Any idea or suggestion? Thanks.

sandeepmistry, 2016-10-27:
@hanabanana how do other boards or networked devices handle this? 

The Atmel ASF code has a `m2m_wifi_set_device_name ` API:

```
/**@}*/
/** @defgroup WifiSetDeviceNameFn m2m_wifi_set_device_name
 *   @ingroup WLANAPI
 *  Set the WINC1500 device name which is to be used as a P2P device name.
 */
 /**@{*/
/*!
@fn			NMI_API sint8 m2m_wifi_set_device_name(uint8 *pu8DeviceName, uint8 u8DeviceNameLength);
@param [in]	pu8DeviceName
			Buffer holding the device name.
@param [in]	u8DeviceNameLength
			Length of the device name. Should not exceed the maximum device name's length M2M_DEVICE_NAME_MAX.
@warning		The function called once after initialization.
@return		The function returns @ref M2M_SUCCESS for successful operations and a negative value otherwise.
*/
NMI_API sint8 m2m_wifi_set_device_name(uint8 *pu8DeviceName, uint8 u8DeviceNameLength);
```

I'm unsure if this is actually used in the DHCP request and if it is your DHCP server and DNS server have to work together to store/update the value and map it to an IP address.

We have a Bonjour/MDNS example bundled with this library.

hanabanana, 2016-10-27:
@sandeepmistry  I would guess that when the device sends DHCP request it embeds its expected hostname into the request. The DHCP server and DNS server work together to see if the name can be used and then store/update it in the map. Specifying a name for a device instead of an IP is much easier to manage for connect/reconnect with DHCP. 

The ESP8266 WiFi library has an API WiFi.hostname which does the exact thing without the need of mDNS and I am a bit surprised that WiFi101 does not have the same support.

sandeepmistry, 2016-10-27:
@hanabanana ok, thanks for the clarification.

We can see if the `m2m_wifi_set_device_name` or another ASF API does this. Would you be comfortable trying to add this call inside the WiFi101 library and seeing if the value is put in the DHCP request?

sandeepmistry, 2016-10-27:
I've quickly tried it out `m2m_wifi_set_device_name` does not set the value of the DHCP hostname.

Here's a screenshot of Wireshark displays:

[image]

sandeepmistry, 2016-10-27:
Unfortunately I don't think we'll be able to support this request without a firmware + ASF code change from Atmel.

The Host Name sent in the DHCP request seems to be fixed at "WINC-xx-yy" where, xx and yy are the last two bytes of the WINC1500's MAC address. `m2m_wifi_set_device_name` is only for setting the device name in P2P mode.

sandeepmistry, 2017-03-13:
Closed via #148 (https://github.com/arduino-libraries/WiFi101/pull/148/commits/9881f995c0c56da9f082ef09bf35d93832782835).

closed 2017-03-13 by sandeepmistry