arduino-libraries/WiFi101 #114: OTA Sketch Upload

LRagji, 2016-11-03:
This is an Enhancement for sure and not a defect, Need some way to do an OTA Sketch Upload as supported by ESP8266 #

sandeepmistry, 2016-11-07:
Hi @LRagji,

Which boards would you like to use this on? At this point, it will require a custom bootloader per board that supports the WINC1500, so it's not likely going to be supported. The [Yun Shield](https://www.arduino.cc/en/Main/ArduinoYunShield) has this feature, however it has a lot more resources to support this.

LRagji, 2016-11-15:
@sandeepmistry  MKR1000

LRagji, 2016-11-29:
Any possibilities for this to happen in near future?

sandeepmistry, 2016-11-29:
Hi @LRagji,

Your request is very interesting but requires a lot of effort. We cannot accomplish this task now, but we will make a note to consider it in the future enhancement pipeline.

ericbaril72, 2016-12-13:
I have done it but my setup uses an SD Card .... which gives me access to the full memory length ( instead of 1/2 ).

However it compiles on the Atmel Studio platform ... I ported the SAMD20 sd bootloader.

ericbaril72, 2016-12-13:
The AP_SimpleWebServer sketch needs 26000 Bytes.
OTA bootloading needs "half" the memory space ( current APP + newVersion + bootloader )

sandeepmistry, 2016-12-13:
@ericbaril72 do you have a link to the (bootloader) code you are using?

ericbaril72, 2016-12-13:
The bootloader "main" stuff is in spi_master_bootloader.c

I still need to "clean it up" to remove as much as possible the heavy ASF / XPLAINED_PRO architecture and keep the BL small and tidy

https://github.com/ericbaril72/SAMD21-SDBootloader

ericbaril72, 2016-12-13:
@sandeepmistry ,
I've been following your work on the ZERO/MKR1000/SAMD architecture for a while and would be more then happy to contribute for the Arduino platform.

I still feel like a N00b is regards to understanding the "Arduino" direction in regards to long term evolution.
It changed dramaticly with the arrival of the DUE / SAMD / ESP8266 and would like some "Yoda guidance" to the Padawan that I am.

I think this bootloader would be usefull for the Arduino platform but to keep the original bootloader as well is better.  This bootloader STILL requires a manual handling of the SD card ( painfull vs serial/USB connection ).   

I Use the WebServer.Ino base sketch to upload files on the SDCard ( and keep the bootloader to a small form factor ).  But ANY screw-up on the "main APP" and its back to manual SDcard swapping .

Merging SDbootloader + WifiClient could easily become a 40-50KB bootloader ... outch !

sandeepmistry, 2017-04-03:
Hi Everyone,

We've released a new [WiFi101OTA library](https://github.com/arduino-libraries/WiFi101OTA) with support for this feature. It requires the latest 1.6.13 release of the SAMD core - which also included support for updating sketches using SD cards with the [SDU](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SDU) library.

https://www.arduino.cc/en/Reference/WiFi101OTA

closed 2017-04-03 by sandeepmistry