arduino-libraries/WiFi101 #115: Using Adafruit Feather M0/Winc1500 loses data on GET

JimWatV1, 2016-11-21:
I have the Feather M0 IoT kit and it works fine for the temp/humidity sample. I tried to use the WiFi101->WiFiWebClient sample and it connects to Wifi ok and gets part of the page, but not the entire page. It will get 3-7K of the google result in the sample. Not a consistent number of bytes.  (I did call SetPins per Adafruit's doc at https://learn.adafruit.com/adafruit-feather-m0-wifi-atwinc1500/using-the-wifi-module)

I tried reading data after disconnect and removing Serial.write and I could get more data, but never the full page. I tried other web pages with same results, partial data. Any ideas on getting the sample to work?

sandeepmistry, 2016-11-24:
Hi @JimWatV1,

Could you please share a URL (and entire sketch that causes issues)?

For the google result sample, I get the following which is the whole page:
```
Starting connection to server...
connected to server
HTTP/1.1 302 Found
Location: http://www.google.ca/search?q=arduino&gws_rd=cr&ei=W2w3WPezPMHgjwSgrq-gCw
Cache-Control: private
Content-Type: text/html; charset=UTF-8
P3P: CP="This is not a P3P policy! See https://www.google.com/support/accounts/answer/151657?hl=en for more info."
Date: Thu, 24 Nov 2016 22:40:27 GMT
Server: gws
Content-Length: 278
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
Set-Cookie: NID=91=zs-DnN3FdI3FqVgTpgiWtokBbQfbRLMLNHktnz9fUNiqG8KSILGCIkQUer1TuUB04kRFoaC2xsiEdj3gDUGR-V37rRoGmcYPU-emwC_Zf2G8NxnihxSCnl9T4K20x6l0; expires=Fri, 26-May-2017 22:40:27 GMT; path=/; domain=.google.com; HttpOnly
Connection: close

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.ca/search?q=arduino&amp;gws_rd=cr&amp;ei=W2w3WPezPMHgjwSgrq-gCw">here</A>.
</BODY></HTML>

disconnecting from server.
````

JimWatV1, 2016-11-28:
Thanks for the reply.  You are getting a 302 to redirect.  Try a link that doesn’t redirect (gets http 200)

I’ve attached the ino I used.  It’s the sample with the one line added, per Adafruit:

  WiFi.setPins(8,7,4,2);

When running it here’s what I get back:  If you hit www.google.com/search?q=arduino<http://www.google.com/search?q=arduino> with Postman or ARC and get a 200 result, you’ll get back much more data.  Notice no closing </body> or </html>

---------------------------------------
Attempting to connect to SSID: Abbey
Connected to wifi
SSID: Abbey
IP Address: 192.168.1.107
signal strength (RSSI):-54 dBm

Starting connection to server...
connected to server
HTTP/1.1 200 OK
Date: Mon, 28 Nov 2016 13:31:07 GMT
Expires: -1
Cache-Control: private, max-age=0
Content-Type: text/html; charset=ISO-8859-1
P3P: CP="This is not a P3P policy! See https://www.google.com/support/accounts/answer/151657?hl=en for more info."
Server: gws
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
Set-Cookie: NID=91=iG8-QSe6TdVSxJTp9cQ6Kb9AU717aRGJRWp6_qouZ6iiacKbb0mSjJRdXwEsR0VtIo8E5xzhWPY1S9bwYLiRzfl6IlSJoRfToM1UV4pj9arkAkNAR95F8F3QBVQcTNoimJmfU-OUV432Kj0; expires=Tue, 30-May-2017 13:31:07 GMT; path=/; domain=.google.com; HttpOnly
Accept-Ranges: none
Vary: Accept-Encoding
Connection: close

<!doctype html><html itemscope="" itemtype="http://schema.org/SearchResultsPage" lang="en"><head><meta content="text/html; charset=UTF-8" http-equiv="Content-Type"><meta content="/images/branding/googleg/1x/googleg_standard_color_128dp.png" itemprop="image"><link href="/images/branding/product/ico/googleg_lodp.ico" rel="shortcut icon"><noscript><meta content="0;url=/search?q=arduino&amp;ie=UTF-8&amp;gbv=1&amp;sei=mzE8WNCpHsWymQGbkIGwCw" http-equiv="refresh"><style>table,div,span,p{display:none}</style><div style="display:block">Please click <a href="/search?q=arduino&amp;ie=UTF-8&amp;gbv=1&amp;sei=mzE8WNCpHsWymQGbkIGwCw">here</a> if you are not redirected within a few seconds.</div></noscript><title>arduino - Google Search</title><style>#gb{font:ght:30px}#gbz,#gbg{position:abacity:1;*top:-2px;*left:-5px;*k;border-top:2px solid transpa.gstatic.com/gb/images/b_8d5afx 4px rgba(0,0,0,.12);-o-box-space:nowrap}.gbmpala{padding-lfbb-hvr{-webkit-box-shadow:0 1bkit-box-shadow:inset 0 1px 2p-image:-ms-linear-gradient(top-stop(.5,rgba(0,0,0,.8)),color0;height:4px}.gbsb .gbsbb:afte:3px;padding-left:8px;text-indgssb_c{left:132px !important;rlor:#808080}._pJb{color:#093}holor:#dd4b39}#mss{margin:.33emn:relative}._nBb>div{outline-cx;margin:0;padding:3px 6px 2pxpx rgba(0,0,0,0.3);-webkit-box&&(!a.getAttribute||!(b=a.getAoogle.dcl=!0;for(var a;a=b.shia){return a in y},A={},B=funct.push(["gl",{url:"//ssl.gstati=function(c){var d=arguments;gn=m.getAttribute("aria-owner")=0;d<c;d++)if(I(a,b[d]))returnocumentElement[a]:0},rb=functiogle.com/u/0",loadTime:(new Da="");return"string"==typeof a?.spl(a,b,"nap",c)}},dc= functiExp("\\b"+ b+"-([0-9]+):"));if"ssp")}g=a},m=e.qs,n=function(",google.log("hpam","&ved="+f.=gbt><a onclick=gbar.logger.iln&tbo=u&tbm=bks&source=og&tab="https://accounts.google.com/Slass="sfbgg" colspan="2" valig">Videos</a></div><div class=" id="tbd"><li><ul class="tbt">bout 68,000,000 results</div><2.4em;overflow:hidden"><a hrefABAF&amp;sig=AOD64_0aKpMZ-QJYDf="/aclk?sa=l&amp;ai=DChcSEwjxww&amp;sa=X&amp;ved=0ahUKEwiQ4tIALYQjBAIMTAC&amp;usg=AFQjCNFsp;- Arduino Zero - Arduin><br></div></div><div class="lass="r"><a href="/search?q=aro the <b>Arduino</b> ...<DG5qpMWIKXjQhXwjmJ7lhtww">Cachfun.com/tutorials/what-is-an-aPCrV0wMRIJ:https://www.adafruie=UTF-8&amp;q=arduino+price+inamp;ei=mzE8WNCpHsWymQGbkIGwCw&vnsl&amp;ei=mzE8WNCpHsWymQGbkIble><script type="text/javascrhtml>
disconnecting from server.

sandeepmistry, 2016-11-28:
@JimWatV1 I can't see the sketch you attached on Github.

Postman and ARC are most likely following the 302 redirect, something the example doesn't do.

I was able to reproduce, by manually updating the sketch for follow the 302 redirect. If I apply the patch from #116, things seem ok.

Can you please try the change from #116 as well?

JimWatV1, 2016-11-29:
Thank you.  Applying the patch to change the MTU fixed it!

sandeepmistry, 2016-11-29:
@JimWatV1 thanks for trying it out!

We've merged #116, and a new v0.11.1 release will be available shortly with the fix.

closed 2016-11-29 by sandeepmistry