arduino-libraries/WiFi101 #119: WiFi.getTime() is returning 1 day older day on MKR1000

LRagji, 2016-12-08:
Found a interesting defect WiFi.getTime() is returning  one day previous to actual date, another thing i could not find is where are we making set time call (m2m_wifi_set_sytem_time) so that i can check if the source is returning correct time.?

Any help would be appreciated on MKR1000 testing with help of http://www.epochconverter.com/ site and RTCZero lib

sandeepmistry, 2016-12-09:
Hi @LRagji,

I've submitted pull request #120 to fix this, could you please try out the change?

If all is good, it will be included in the next WiFi101 library release.

LRagji, 2016-12-09:
@sandeepmistry This works with MKR1000 just tested it Thanks

closed 2016-12-09 by LRagji

sandeepmistry, 2016-12-09:
@LRagji thanks for trying it out, I've merged the fix in #120, so it will part of the next WiFi101 library release.

sandeepmistry, 2017-01-05:
v0.12.0 has been released today with the latest changes from master, which includes the fix.

PR #120 (2016-12-09): Correct WiFi.getTime() return value off by one day
`tm.tm_mday` expects a range of 1 - 31, but the WINC uses 0 - 30 for `u8Day`.

Fixes #119.