arduino-libraries/WiFi101 #121: WiFi.RSSI() causing lock up on MKR1000

facchinm, 2016-12-15:
_Moved from https://github.com/arduino/Arduino/issues/5725_ @tonywhitfort 

The following code demonstrates WiFi.RSSI() causing the processor to lock up.
In this example the lockup is demonstrated by the LED which stops flashing after a few seconds or sometimes a bit longer after a restart.
I am using MKR1000 with the WiFi101 library.

I have found that not calling WiFi.RSSI() for 20secs after WiFi.begin() seems to fix the problem
Shorter delays cause varying degrees of unreliability, the shorter the worse.
See the code comments for more detail

Sorry for the lack of code formatting - I tried using the insert code feature but it didn't seem to have the desired affect - not in the Preview tab away. What am I missing?

```C++
#include <WiFi101.h>

char ssid[] = "xx"; 
char pass[] = "xx";   

#define SECS  2
// for SECS = 2  - very unreliable - locks up after a few seconds more often than not
// for SECS = 10 - more reliable   - locks up on every 5th to 10th restart
// for SECS = 20 - seems reliable

void setup() 
{ 
  pinMode(LED_BUILTIN, OUTPUT);
  
  int status = WL_IDLE_STATUS;  
  while (status != WL_CONNECTED) {  
    status = WiFi.begin(ssid, pass);
  }
}

void loop() 
{
  static int x = 0;

  if (millis()/1000 < SECS) { 
    digitalWrite(LED_BUILTIN, x++ % 5 ? HIGH : LOW); // slow flash   
  } else {   
    WiFi.RSSI();
    digitalWrite(LED_BUILTIN, x++ % 2 ? HIGH : LOW); // fast flash to identify when delay finished
  }
  delay(100);
}
```

sandeepmistry, 2016-12-15:
Hi @tonywhitfort,

I've submitted pull request #122 to fix this issue. It would be great if you had time to try out the change before we merge it. Thanks.

closed 2016-12-15 by facchinm

tonywhitfort, 2016-12-15:
Yep Fixed Thanks :)

PR #122 (2016-12-15): Fix crashing issues with RSSI
Fixes #121.

cc/ @facchinm @agdl @akash73