arduino-libraries/WiFi101 #124: No example for the WiFi.getTime() function

drp0, 2016-12-17:
The wifi101 library offers WiFi.getTime() to return the epoch time.
This can be used with the avr-libc time.h library
and **not** with the <TimeLib.h> library
 Reference [https://github.com/arduino-libraries/WiFi101/issues/111]

After a lot of digging I found a mac example demonstrating the time.h library functions
I have adapted the code to work with the wifi101
The library is a bit cumbersome to work with and one of the key functions to produce a sdfat timestamp appears not to be correctly linked in the core.
Despite this, the library has a lot of very powerful esoteric functions.
``` C++
// Timetest.ino

// adapted from Mac example at http://swfltek.com/arduino/builds/1.6.6/

// useful Michael Duane <time.h> source listing http://www.nongnu.org/avr-libc/user-manual/time_8h_source.html
// time manual http://www.nongnu.org/avr-libc/user-manual/group__avr__time.html

// D.R.Patterson
// 17/12/2016

// Tested on a AtMega 2560 

// The time library time.h can be initialsed with an epoch time
// this can be entered from a Serial interface, or
// if WiFi is available the initial time can be set with a call to WiFi.getTime()

// for an led pulse the code requires:
  // two pins with a 1Kohm resistor in series with an led
  // the led cathode grounded to irqpinlo
  // the resistor in series with the anode, connected to irqpin
  
// To maintain the clock on the Time.h library the function system_tick() must be
// called at one second intervals.
// TimerOne.h library facilitates this
// This has less overhead than repeatedly calling the WiFi.getTime() function
  
// The code uses "double" for many non-integer variables
// On the Uno and other ATMEGA arduino this is interpreted as float (4-bytes)
// The Arduino Due interprets double as 8-bytes (64 bit) precision

#include <time.h>
//#include <util/usa_dst.h> // usa daylight saving
#include <util/eu_dst.h>    // eu daylight saving
#include <math.h>
#include <time.h>
#include <TimerOne.h>

#define RAD_TO_DEGREE 57.29577951308233
#define BTRA 5.91944444444444
#define BTDECL 7.4071

#define MY_LATITUDE 54.99065 * ONE_DEGREE   // my latitude (degrees) is North, therefore +
#define MY_LONGITUDE -1.59988 * ONE_DEGREE  // my longitude is east therefore -
#define MY_TIME_ZONE 0 * ONE_HOUR           // gmt offset 0
  // The 'time zone'  parameter is given in seconds East of the Prime Meridian.
  // For New York City: -5 * ONE_HOUR
//#define MY_DST usa_dst                    // usa time saving
#define MY_DST eu_dst                       // eu  daylight time saving

// buffer and index to store incoming serial data
char db[19];
int dbi=0;
volatile boolean toggle = false;  // led on off flag
volatile boolean badirq = true;   // signal 1st irq trigger
boolean isSet = false;
boolean clockset=false;

const byte pinirq = 26;   // irq led pulse
const byte pinirqlo = 28; // irq led ground

[... 339 lines cut ...]
// pass control to AltAz()
AltAz(HourAngle, Decl / RAD_TO_DEGREE, Azimuth, Altitude);
}

void MoonPosition(time_t * point_in_time, double * Azimuth, double * Altitude){
// at solar noon at new moon, moons ha is same as suns
// the ha 'elongates' as moon progresses to full

time_t noon = solar_noon(point_in_time) % ONE_DAY;// time of noon mod one day
time_t tday = *point_in_time % ONE_DAY; // current time mod one day
long r = tday - noon; // offset from noon, -43200 to + 43200 seconds
// r = HA of sun in seconds

long elongation =moon_phase(point_in_time) * -432L;

r = r + elongation;
 
double HourAngle = r / 86400.0; // hour angle as percentage of full day
HourAngle = HourAngle * 2.0 * M_PI; // hour angle in radians

double Declination = solar_declination(point_in_time);// declination in radians
AltAz( HourAngle, Declination,Azimuth,Altitude );
}
// *****************End Advanced time libray functions *****************
```

sandeepmistry, 2016-12-23:
@drp0 where you thinking something like this?

```arduino
/*

  Udp NTP Client

  Get the time from a Network Time Protocol (NTP) time server
  Demonstrates use of UDP sendPacket and ReceivePacket
  For more on NTP time servers and the messages needed to communicate with them,
  see http://en.wikipedia.org/wiki/Network_Time_Protocol

  created 4 Sep 2010
  by Michael Margolis
  modified 9 Apr 2012
  by Tom Igoe

  This code is in the public domain.

*/

#include <SPI.h>
#include <WiFi101.h>

int status = WL_IDLE_STATUS;
char ssid[] = "mynetwork";  //  your network SSID (name)
char pass[] = "mypassword";       // your network password
int keyIndex = 0;            // your network key Index number (needed only for WEP)

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  // attempt to connect to Wifi network:
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }

  Serial.println("Connected to wifi");
  printWifiStatus();
}

void loop() {
  unsigned long epoch = WiFi.getTime();

  if (epoch > 0) {
    // now convert NTP time into everyday time:
    Serial.print("Unix time = ");
    // print Unix time:
    Serial.println(epoch);

    // print the hour, minute and second:
    Serial.print("The UTC time is ");       // UTC is the time at Greenwich Meridian (GMT)
    Serial.print((epoch  % 86400L) / 3600); // print the hour (86400 equals secs per day)
    Serial.print(':');
    if ( ((epoch % 3600) / 60) < 10 ) {
      // In the first 10 minutes of each hour, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.print((epoch  % 3600) / 60); // print the minute (3600 equals secs per minute)
    Serial.print(':');
    if ( (epoch % 60) < 10 ) {
      // In the first 10 seconds of each minute, we'll want a leading '0'
      Serial.print('0');
    }
    Serial.println(epoch % 60); // print the second
  }
  // wait ten seconds before asking for the time again
  delay(10000);
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

I've modified the [WiFiUdpNtpClient](https://github.com/arduino-libraries/WiFi101/blob/master/examples/WiFiUdpNtpClient/WiFiUdpNtpClient.ino) example.

drp0, 2016-12-24:
Having examined the time library at some length, there is no need to extract the time by calculation-
The library does all that and more (see above)
I suggest that end users will want:

1) To ignore epoch values of zero-
see **if (epoch == 0) continue;** section above

(Thought: it may be worth increasing the timeout value for WiFiClass::getTime() )
 
2) a simple method of getting the time:
``` C++
#define MY_TIME_ZONE 0 * ONE_HOUR           // gmt offset 0
  // The 'time zone'  parameter is given in seconds East of the Prime Meridian.
  // For New York City: -5 * ONE_HOUR
//#define MY_DST usa_dst                    // usa time saving
#define MY_DST eu_dst                       // eu  daylight time saving

set_dst(MY_DST);        // eu daylight saving in my case
set_zone(MY_TIME_ZONE); // GMT 0 in my case

set_system_time( (time_t)epoch ); 
time(&tnow); // store time in variable tnow

gmtime_r(&tnow, &gm);          // convert tnow and place GMT in gm time pointer
localtime_r(&tnow, &lt);       // place local time in lt
String ascTime = asctime(&lt); // day and month as text, based on local time
```
3) A method of parsing the time structure-
see **void showdetails**

4) A method of incrementing the time, since the library does not do it automatically
see timeone irq above
It can take up to 60 seconds to get a valid epoch time, so repeated wifi calls are not useful!

5) A call back function to set sdcard file timestamps
see **void dateTime**

Having chosen to base epoch time on library time.h in wifi101, good examples of the use of time.h should be given.
There are no examples of time.h in the arduino release package.
I am happy for you to use my examples in whole, or part.

David

sandeepmistry, 2016-12-29:
@drp0 right, but `time.h` is not part of the Arduino libraries, it is bundled with the GCC toolchain, and part of the POSIX API.

drp0, 2017-01-03:
YiFi101 includes time.h in WiFi.cpp
Having done so, there should be usable examples as detailed above.