arduino-libraries/WiFi101 #125: WiFi.getTime returning bad values at startup

tonywhitfort, 2016-12-18:
The WiFi.getTime function returns bad values for a period of time after WiFi.begin.
I expected it to return 0 during the initial period until it had a valid time.

In my case it always reports 2254611840 as the first value and often keeps reporting that same value for a period of 2 to 15 seconds then it starts reporting a correct value that increments each second.

The example code below prints the returned value each second after startup.

Typically the output to the Serial Monitor will show something like.
2254611840
2254611840
2254611840
1482028206
1482028207
1482028208
1482028209
1482028210

`#include <WiFi101.h>

char ssid[] = "xx";
char pass[] = "xx";

void setup()
{ 
  int status = WL_IDLE_STATUS;
  while (status != WL_CONNECTED) {
    status = WiFi.begin(ssid, pass);
  }
}

void loop()
{
  Serial.println(String(WiFi.getTime()));
  
  delay(1000);
}
`

drp0, 2016-12-18:
I have observed that the first two values returned were
1313798640
1313798640
These correspond to the year 2041
I am now rejecting any returned date with the year > 2020
It would be very useful to have WiFi.getTime return an error value if the epoch time was not available.
``` C++
    for(byte i = 0 ; i < 5; i++){
    unsigned long epoch = WiFi.getTime(); // get wifi time
    set_system_time( (time_t)epoch );     // set the time library
    time_t tnow;
    struct tm lt;
    time(&tnow); // store time in variable tnow
    localtime_r(&tnow, &lt);       // place local time in lt
    unsigned int date = getyear(&lt);
      if(date < 2021) i = 5; else delay(5000);
    }

unsigned int getyear(const struct tm * timeptr) {
return timeptr->tm_year + 1900;
}

```
David

drp0, 2016-12-18:
I have had a look at lines 951 to 971 of WiFi.cpp
systemTime.u16Year is returning 0 (presumably) when the time is not available.
When 1900 is subtracted this produces a **tm.tm_year= -1900**
I suspect, but have not tested, that **tm.tm_mon = -1**
and
**tm.tm_mday = +1**
may also occur.
My solution:
``` C++
	if (_resolve == 0) {
		struct tm tm;
		unsigned long tt = systemTime.u16Year;	// drp
		Serial.println(tt);					// drp for testing
			if(tt == 0) return 0;				// drp
		tm.tm_year = tt - 1900;				// drp
		//tm.tm_year = systemTime.u16Year - 1900;	// drp
		tm.tm_mon = systemTime.u8Month - 1;
		tm.tm_mday = systemTime.u8Day + 1; 
		tm.tm_hour = systemTime.u8Hour;
		tm.tm_min = systemTime.u8Minute;
		tm.tm_sec = systemTime.u8Second;
		tm.tm_isdst = -1;
		t = mktime(&tm);
		return t;
	} else { 								// drp
	_resolve = 0;							// drp
	return 0;								// drp
	}									// drp
	//_resolve = 0;                                                  // drp

	//if (t == (time_t)-1) t = 0;				// drp

	//return t;	                        				// drp
```
David

tonywhitfort, 2016-12-18:
Thanks David
That works for me.
I now get 0 returned until the correct time is available

sandeepmistry, 2016-12-23:
Hi @tonywhitfort, @drp0,

Thanks for letting us know, I opened pull request #126 to resolve this. `WiFi.getTime()` will return `0` when the WINC1500 has not yet sync the NTP time.

Could you please try out theP PR #126 and provide your feedback?

tonywhitfort, 2016-12-24:
Yes this works for me
Thank you

drp0, 2016-12-24:
re [https://github.com/arduino-libraries/WiFi101/pull/126/commits/825dac1cdcaa09ddcbc230239d7aaec6341eb203](https://github.com/arduino-libraries/WiFi101/pull/126/commits/825dac1cdcaa09ddcbc230239d7aaec6341eb203)

Unless it has ramifications elsewhere.?.
at line 989 why set t to a negative number and then return uint32_t
Set it to zero, then code becomes
````C++
	time_t t = 0;

	if (_resolve == 0 && systemTime.u16Year > 0) {
		struct tm tm;

		tm.tm_year = systemTime.u16Year - 1900;
		tm.tm_mon = systemTime.u8Month - 1;
		tm.tm_mday = systemTime.u8Day + 1;
		tm.tm_hour = systemTime.u8Hour;
		tm.tm_min = systemTime.u8Minute;
		tm.tm_sec = systemTime.u8Second;
		tm.tm_isdst = -1;

		t = mktime(&tm);
	}else 	_resolve = 0;

   return t;
}
''''

closed 2016-12-29 by agdl

sandeepmistry, 2016-12-29:
@tonywhitfort thanks for testing the change!

@drp0 I've added your suggestion in https://github.com/arduino-libraries/WiFi101/commit/1ebdfacea34bc42c970dc37d4bd017ac273a8fbe, thanks as well!

sandeepmistry, 2017-01-05:
v0.12.0 has been released today with the latest changes from master, which includes the fix.

PR #126 (2016-12-29): Make WiFi.getTime() return 0 if WINC1500 hasn't synced time
Resolves #125.

cc/ @tonywhitfort @drp0