arduino-libraries/WiFi101 #127: Problem when running WiFiClient and WiFiServer at the same time

sontono, 2016-12-29:
Hi,
I am running library on MKR1000 with IP 192.168.0.37.
I start a WiFiServer (ws). Afterwards, I create a WiFiClient (wc) and connect it to an external IP and port (192.168.0.18:8181).
The problem now is that ws.available() returns the client wc as it was a client connected to the own server.
This is the full code:

`#include <WiFi101.h>

WiFiClient wc;
WiFiServer ws(5000);

void setup() {
	// Open serial port
	Serial.begin(9600);

	// Connect to wifi
	WiFi.begin("ENXINE DEV", "20548905");
	int i = 0;
	while (WiFi.status() != WL_CONNECTED && i < 5) {   
		delay(500);
		Serial.print(".");
		i++;
	}
	delay(100);
	if (WiFi.status() != WL_CONNECTED) {
		Serial.println(F("Error: WiFi not connected!"));
	}
	
	// Start server
	ws.begin();

	// Connect client
	wc.connect("192.168.0.18", 8181);
	
	// Test
	if (ws.available()) Serial.print("WTF! Client connected!!!. my_id: ");
	else Serial.println("NO client connected!!!");
}

void loop() {
	delay(1000);
}
`
Thanks

sandeepmistry, 2016-12-29:
Hi @sontono,

Thanks for letting us know about this issue, and the excellent issue report!

I've submitted pull request #128 to resolve this, could you please try out the change?

sontono, 2016-12-29:
As commented in #128, that pull request fix this issue

closed 2016-12-29 by sontono

sandeepmistry, 2017-01-05:
v0.12.0 has been released today with the latest changes from master, which includes the fix.

sontono, 2017-01-06:
Great!!. Thank you very much

PR #128 (2016-12-29): Fix WiFiServer::available() returning valid client on outbound client connection
Resolves #127. Sketch to reproduce issue can be found here: https://github.com/arduino-libraries/WiFi101/issues/127#issue-197982939

cc/ @sontono